hifiberry/hifiberry-os #431: Volume slider not adjustable with Spotify Connect

hyperbart, 2023-03-09:
First of all let me say how awesome this OS and the design of it is, I was really impressed with the UX and the integration within the B&O system.

However, there is something that makes my eyelids twitch a bit ;)

**Describe the bug**
When playing music via Spotify Connect the volume slider on the web interface is

1. non responsive,
2. is always at 0%,
3. jumps back to 0% when you slide
4. when it works it is not "in sync" with the device, so the sliders operate stateless of each other.

**HiFiBerryOS version**
[image]

**HiFiBerry sound card**
[image]

**Powersupply**
Meanwell 20V Power Supply 200W

**To Reproduce**
Steps to reproduce the behavior:
1. Enable Spotify Connect
2. Open Spotify on iPhone
3. Play track and play it via Spotify Connect on the Hifiberry
5. Go to the Webinterface of your Hifiberry
6. Try to manipulate the volume via the web interface
7. See error
8. After some reboots and trying again the volume slider just now started working
9. however the volume is not synced across the devices.

**Expected behavior**

1. Volume slider shows current volume
2. Volume slider can be controlled

**Screenshots**

**Browser (if applicable)**
 - OS: iOS
 - Browser Safari
 - OS: Windows 11
 - Browser Firefox

hifiberry, 2023-03-09:
Looks like something is broken in your setup as we don't see this here. What sound card are you using?

hyperbart, 2023-03-09:
It's a fresh install, as in: 2 days old.

[image]

hyperbart, 2023-03-09:
> 

Have you tried syncing the volumes, they don't move with each other, if you change the webplayer volume, the Airplay volume slider does not change. Must be reproducable.

hifiberry, 2023-03-09:
Works here just fine. Is this a fresh installation?

hifiberry, 2023-04-03:
Unfortunately I can't reproduce this here. It's just working fine in several installations here.

closed 2024-01-10 by hifiberry