hifiberry/hifiberry-os #432: DSP13 breaks optical input

bobmoss, 2023-03-15:
**Describe the bug**
Version 13 of the DSP13 stops the optical input working

**HiFiBerryOS version**
20230223

**HiFiBerry sound card**
DAC+ DSP

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Sound
2. Click on DSP Programs
3. Select and install DAC+ DSP Universal version 13
4. Start playing music on a device via the Optical input.
5. Click on Sources
6. Click On Optical Input

Music does not play and on the Optical input page status shows "Not Active".

**Expected behavior**
Music should be playing and Optical input status should show "Active"

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Browser (if applicable)**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Additional context**
Version 12 of the DSP works OK, but shows a warning of "Sampling rate is not defined. Filters can't be controlled".

zajc3w, 2023-03-21:
Same problem here unknown dps program dsp came with worked, but no filters were availiable(obviously).
HiFiBerryOS version
20230223
Rpi3B

optical works with profile 12, but DSP is basically a passtrough due to aforementioned "Sampling rate is not defined. Filters can't be controlled".

is there v 12 and v 13 sigma project available? v11 doesn't work, probably because it was developed fo adau1151??

hifiberry, 2023-04-04:
The updates DSP profile (v14) is now available. Can you  try the latest experimental release with DSP profile v14?

/opt/hifiberry/bin/update --experimental

Rektedekte, 2023-04-04:
Everything is working at my end, at least. Nice to switch over from bluetooth, good work 👍

hifiberry, 2023-04-04:
Thank you for the feedback!

Rektedekte, 2023-04-05:
Perhaps I spoke a bit too soon... A restart later, it no longer works.

I'm starting to wonder if something else is wrong on my end, I had the same behavior when rolling back to v12.1. It works while the hifiberry is on, then never again, no matter how many times I try to reinstall it. The issue is definitely different than the one on v13, the "playing" light triggers exactly as it should, just no sound.

Funny thing, just scrolling through a menu with UI sound causes the hifiberry to emit that low frequency clicking, as if it's constantly enabling and disabling output.

hifiberry, 2023-04-05:
Check in the TOSLink source if a signal is detected.

Rektedekte, 2023-04-05:
It is detected, the "not active" changes to "playing" when sound is being played, just no sound on the output. I have checked that other sources work.

bobmoss, 2023-04-05:
I've just tried this by downloading the latest version (20230404) and can confirm loading the DSP V14 from that is working OK. I've done a couple of reboots ands it still works OK.

Rektedekte, 2023-04-05:
I really do have a unique issue then :(. I've tried all the optical output settings, verified that it does indeed output with other devices, tried a different cable and source, but the problem must be with the hifiberry, cause it worked on the first boot cycle. I suppose i can try reinstalling the OS.

Rektedekte, 2023-04-05:
Did a completely fresh install of HiFiberry OS, updated to experimental and installed v14. No sound from toslink, not even on first install. Just straight to the same problem. I confirmed that the cable is receiving sound via another dac i have. I'm starting to think there's something wrong with mine.

hifiberry, 2023-04-05:
If the UI say that a signal is detected, there is definitely a signal received from the DSP. And if output from the Pi is working fine, the output is also ok. The only thing I can offer is to check your whole setup (Pi+SD card+DAC) here to find out what's wrong in your setup. If you want to do this, you can contact support@hifiberry.com

zodiaq, 2023-04-06:
@Rektedekte I don't know if your issue is the same as mine, but I also need the TOS link output after powering on.
I need to play another source once to get the TOS link output in my case.
For instance, I run "play -n synth 1 sine 400" then I can hear TOSlink input after the playback.

bobmoss, 2023-04-06:
@zodiaq I can confirm I have exactly the same issue! If I power off my device, reboot it and start playing a TOSlink input, it says in HiFiBerry "Status - Playing" for the optical input, but no sound. I then play a different source, stop that source and the optical starts working!

Rektedekte, 2023-04-06:
> play -n synth 1 sine 400

How in the world did you figure that out? Can confirm, fixes it on my end. Though it's odd, playing from spotify connect didn't have the same effect. Thank you very much :)

Rektedekte, 2023-04-06:
A script to play a quiet synth (/root/start.sh):

```sh
#!/bin/sh
play -n synth 1 sine 400 vol 0.0001
```

Service file for systemd (/etc/systemd/system/reboot.service):

```
[Unit]
Description=Reboot systemd service.

[Service]
Type=simple
ExecStart=/bin/bash /root/start.sh

[Install]
WantedBy=multi-user.target
```

Run commands:
> chmod 644 /etc/systemd/system/reboot.service
> systemctl enable reboot.service

This will create a systemd service that runs on startup, triggering a quiet synth playback. Works for me, ensures that toslink works on startup.

flavorplus, 2023-04-06:
Same problem, even DSP14 still breaks the SPDIF input. I'm running the latest OS with DSP12 at this point. That combination works but as soon as I switch to DSP13 or DSP14, audio stops working from the SPDIF

Rektedekte, 2023-04-06:
@flavorplus 
On v14, does it say "playing" or "not active" when sound is playing? Seems to indicate two seperate problems.

flavorplus, 2023-04-06:
Ah okay, it does say playing. It'a receiving a signal but no audio. Audio from other sources still works.

Rektedekte, 2023-04-06:
Only solution i know would be the one already mentioned. SSH in and run:

`play -n synth 1 sine 400 vol 0.0001`

Don't know if you've tried that yet. Others mentioned simply needing to play sound from other sources, but for me, playing from Spotify didn't fix it.

ymetivier, 2023-05-02:
Unable to update DSP profile.
Command '/opt/hifiberry/bin/update --experimental' returns "Couldn't get release information from https://musicdb.hifiberry.com/update-system, aborting...

Rektedekte, 2023-05-03:
Usually happens when the servers are down for maintenance, wouldn't know though :/

ymetivier, 2023-05-08:
OS is uptodate : "you are already using the latest version (20230404 vs. 20230404)"
But no v14 DSP profile?!
How can I downolad it?

zodiaq, 2023-05-17:
@RPhilbo What a great job! This explains why TOS link output comes when I run synth once.

But unfortunately, the versions of the DSP profile earlier than DSP14 have the serious issue to flash the board.
Does your profile contain the fix of that?

hifiberry, 2023-05-17:
@RPhilbo Thank you for this. The SPDIF Lock register bug has been fixed already in v14. We haven't seen the "PiDetect" issue here, but this would explain why it works for some people, but not others. It seems that the ASRC sometimes just repeats the last sample. We have seen this only with SPDIF, but it looks like the same issue exists also on other inputs. In this case an DC blocker should also work here. Is that what you did?

I will check where we can also store the SigmaStudio files somewhere in this repo.

RPhilbo, 2023-05-17:
@zodiaq To be honest, i never had a problem with flashing the board, not with webGUI nor directly out of sigma studio.
(Loosing connection while in sigma studio due to wifi packet loss i think often, but you mentioned "flash the board")

I dont know were i found that, also cant find it right now, but once i read about the "SPI write speed" in Sigma Studio and lowered it from the original 10 MHz to 1 Mhz, maybe thats the reason.

@hifiberry That would be really helpful to have the sigma studio file for every (or from now on every) .xml profile, so others can start from that and just change these little portions.
The DC-blocker is just in front of the Pi signal detection block:
[image]

hifiberry, 2023-05-17:
Thanks. I'll add this to our profile here. 

Yes, the SPI write speed of 10MHz doesn't work with all boards. With 1MHz SPI write speed the profile should work fine on all boards.

hifiberry, 2024-01-19:
The DSP profile v15 supplied with HiFiBerryOS64 fixes this. Feel free to try the pre-release.

closed 2024-01-19 by hifiberry