hifiberry/hifiberry-os #433: DSP13 swaps Left and Right speakers

bobmoss, 2023-03-15:
**Describe the bug**
Version 13 of the DSP swaps the left and right speakers over.

**HiFiBerryOS version**
20230223

**HiFiBerry sound card**
DAC+ DSP

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Sound
2. Click on DSP Programs
3. Select and install DAC+ DSP Universal version 13
4. Start playing music.
5. Click on Sound
6. Click On Channels
7. Move balance slider over to the left.
8. You will notice that the left speaker volume is reduced
9. Move balance slider over to the right.
10. You will notice that the right speaker volume is reduced
11. Set the balance back to center
12. Go to the advanced page
13. Mute speaker A (Left), the right speaker is actually muted.

**Expected behavior**
If you move the balance to the left, the right speaker volume should be reduced.
If you mute Speaker A, the left speaker should be muted.

**Screenshots**

**Browser (if applicable)**

**Additional context**
Version 12 of the DSP works OK, but shows a warning of "Sampling rate is not defined. Filters can't be controlled".

hifiberry, 2023-03-17:
You can fix V12 DSP profile by simply changing "samplerate" to "sampleRate"

There is no fixed left/right mapping to a specific channel. You can route any left/right to any channel. What exactly did you configure? You can see the channel mapping in the advance DSP settings.

hifiberry, 2023-04-04:
I can't reproduce this here. You might try the latest experimental release with DSP profile v14

/opt/hifiberry/bin/update --experimental

jimipj, 2023-09-24:
I stumbled upon a similar problem when I was tinkering with DSP profile v14.

If the profile is installed in the normal way, the channels are correct.
However, if I download the same v14 profile from the dspprofiles repo and apply it without any changes using SigmaStudio, the channels are swapped.

This was also with DAC+ DSP.