hifiberry/hifiberry-os #434: NFS

jzahraoui, 2023-03-22:
DO NOT remove the blocks below, but fill these with the requested data. Incomplete feature requests will be closed.
You should remove this comment before posting the feature request.

**Is your feature request related to a problem? Please describe.**
try to access to my nfs server.
Have my music files on a linux server.
hifiberry is also on linux, the natural way to access it is nsf.
mpd can natively access to nfs.

**Describe the solution you'd like**
First I tried to mount the nfs file system, but neet to install nfs-utils, can you tell me hos to do it please ?

```
mkdir /data/library/music/nfs/
mount -t nfs $IP:$SHARE /data/library/music/nfs/
mount: /data/library/music/nfs: bad option; for several filesystems (e.g. nfs, cifs) you might need a /sbin/mount.<type> helper program.
```

second way was to do it through mpd

```
vi /etc/mpd.conf
# add this line :
music_directory         "nfs://$IP/$SHARE"
```

then do this :

```
mpc update
mpc ls
```

but the web interface return a lot of errors messages :

```
(node:16543) UnhandledPromiseRejectionWarning: TypeError: Cannot read property 'Metallica' of undefined
    at Object.getMusic (/opt/beocreate/beo-extensions/mpd/index.js:518:20)
    at /opt/beocreate/beo-extensions/music/index.js:192:47
    at Array.forEach (<anonymous>)
    at getFromMultipleProviders (/opt/beocreate/beo-extensions/music/index.js:190:12)
    at EventEmitter.<anonymous> (/opt/beocreate/beo-extensions/music/index.js:107:5)
    at EventEmitter.emit (/usr/lib/node_modules/eventemitter3/index.js:202:33)
    at BeoCom.<anonymous> (/opt/beocreate/beo-system/beo-server.js:994:10)
    at BeoCom.emit (events.js:400:28)
    at WebSocketConnection.<anonymous> (/opt/beocreate/beocreate_essentials/communication.js:132:12)
    at WebSocketConnection.emit (events.js:400:28)
(Use `node --trace-warnings ...` to show where the warning was created)
UnhandledPromiseRejectionWarning: TypeError: Cannot read property 'Metallica' of undefined
    at Object.getMusic (/opt/beocreate/beo-extensions/mpd/index.js:518:20)
    at /opt/beocreate/beo-extensions/music/index.js:192:47
    at Array.forEach (<anonymous>)
    at getFromMultipleProviders (/opt/beocreate/beo-extensions/music/index.js:190:12)
    at EventEmitter.<anonymous> (/opt/beocreate/beo-extensions/music/index.js:107:5)
    at EventEmitter.emit (/usr/lib/node_modules/eventemitter3/index.js:202:33)
    at BeoCom.<anonymous> (/opt/beocreate/beo-system/beo-server.js:994:10)
    at BeoCom.emit (events.js:400:28)
    at WebSocketConnection.<anonymous> (/opt/beocreate/beocreate_essentials/communication.js:132:12)
    at WebSocketConnection.emit (events.js:400:28)
    at emitUnhandledRejectionWarning (internal/process/promises.js:168:15)
    at processPromiseRejections (internal/process/promises.js:247:11)
    at processTicksAndRejections (internal/process/task_queues.js:96:32)
(node:16543) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 2)

```

**Describe alternatives you've considered**
Use the samba but not makes me happy at all.

**Additional context**
from a clean install.

hifiberry, 2023-03-22:
No NFS support is planned.

jzahraoui, 2023-03-22:
this is what I read into other issue.
will be great if I find a solution to get it work.
there a procedure to install packages without any package manager ?

pinballpower, 2023-03-22:
You have to build your own release for this. You can't just add some packages.

jzahraoui, 2023-03-22:
thank you, this is what I thought.
nfs through MPD show the album under music. but if you click on one of them, it's empty.
but if you make a search it will find the tracks and the album. and most important it is able to play it.
I think there is very close, simply an update to the music package will do the stuff.
it seam to be not updated since years : [beocreate-music](https://github.com/tuomashamalainen/beocreate-music/commits/development)

jzahraoui, 2023-03-22:
happy to ear (or read) this 👍 
tried with the command line mpc all is showed (albums, artists, tracks, etc.) on the hifiberry itself.
also installed this project on a remote server [cyp](https://github.com/ondras/cyp)
I choose this one because it is in nodejs like music extension
accessed via http://<REMOTE_IP>:8080/?server=hifiberry.local:6600
all is working fine :)

hifiberry, 2023-03-24:
No WebDAV planned. If you want to access a file server, SMB it is. No other protocols are planned.

closed 2023-03-24 by hifiberry