hifiberry/hifiberry-os #436: Cant Build own Image (invisible-mirror.net) not connecting

LopeKinz, 2023-03-23:
**Describe the bug**
While building my own image (invisible-mirror.net) is not connecting. (Cant proceed)

**HiFiBerryOS version**
20230223

**HiFiBerry sound card**
None

**To Reproduce**
```bash
bash build-all 
```

**Expected behavior**
Building Hifiberry lol

**Screenshots**
https://ibb.co/7kXWMbb

**Browser (if applicable)**
None

**Additional context**
building on windows with ubuntu subsystem

pinballpower, 2023-03-23:
Sometimes external servers hosting packages can be down for some time. We can't do anything about this. Try again later. If it still doesn't work, check if your ISP might be doing some URL/DNS blocking.

hifiberry, 2023-04-03:
External servers can be unreachable sometimes. Try again later. If it always happens, check with our ISP.

closed 2023-04-03 by hifiberry

LopeKinz, 2023-04-03:
> External servers can be unreachable sometimes. Try again later. If it always happens, check with our ISP.

but the servers are online and i can download the file manually. but the wget command doesnt seem to download the file.

is there a way to manually add the needed file?