hifiberry/hifiberry-os #443: audiocontrol2 plugin gets deleted on OS update

borkog, 2023-04-13:
I wrote an ac2 plugin, it gets deleted when I update os. 

Plugin is located in /opt/audiocontrol2/ac2/plugins/control/

Not sure whether to call this a bug... Is there a workaround?

LopeKinz, 2023-04-17:
its because the os itself replaces alot of files. so your plugin gets deleted due to the change of some files from audiocontrol. so its not a bug

hifiberry, 2023-04-17:
@LopeKinz is (partially) right. You should not put any plugins in /opt. The only safe file system that will survive an update is /data.

Have a look here:
https://github.com/hifiberry/audiocontrol2/blob/master/doc/extensions.md

closed 2023-04-17 by hifiberry

borkog, 2023-10-17:
thanks ... but where do I put

plugin_dir=/data/ac2plugins
metadata=MyDisplay