hifiberry/hifiberry-os #444: Folder View Nas Sources.

saturnson, 2023-04-14:
**Is your feature request related to a problem? Please describe.**
When listing the music on my NAS from within the gui of HiFiBerryOS I can only see the artist and album view. Please could you enable folder view. People don't want to have to go through their whole music collection and change the metadata to suite HIFIBerry OS. When looking for an album that is a collaboration for instance if the folder view was implemented it would make finding the album a lot easier.

**Describe the solution you'd like**
Have a folder view tab in the music sources section.

**Describe alternatives you've considered**
I have used moode audio, picoreplayer and volumio and prefer the clean interface and audio quality of hifiberryOS but dislike that the library system is a mess. moode audio is on another sd card, but I would like to stick with hifiberry if you can implement folder view.

**Additional context**
Also work on implementing 64bit to make the OS quicker as it is great as a renderer but as a snappy os with a web interface it is not as good as it could be.

hifiberry, 2023-04-14:
There are no plans for a "folder view". When you tag your files correctly with Artist, title, album artists and album title, there the library system is working well. 

The performance of the web browser has nothing to do with 32 vs 64bit, but with limitations that we have on available web browsers. There might be a way to use another web browser in the future, be we can't say if and when this might happen.

closed 2023-04-14 by hifiberry

saturnson, 2023-04-14:
That's a pain as I have quite a large library and it is impossible to correct all of the metadata so that hifiberry is happy. I am sure that I am not the only user who would like to see this implemented. I love the software but it is unusable with its current library setup when indexing from a nas.

hifiberry, 2023-04-14:
In this case you need to stick with one of the other distributions. 

BTW: There are good tagging tools around:
https://www.hifiberry.com/blog/tools-to-tag-your-music-collection/

saturnson, 2023-04-16:
Not a great solution as I have a lot of music and bootlegs. I'll have to go back to moode, shame.