hifiberry/hifiberry-os #456: Ability to send CR (Carriage Return 0D) instead of LF (Line Feed - 0A) after serial interaction

chazman88, 2023-05-10:
DO NOT remove the blocks below, but fill these with the requested data. Incomplete feature requests will be closed.
You should remove this comment before posting the feature request.

**Is your feature request related to a problem? Please describe.**
When creating serial/RS232 interactions it appears that HifiberryOS always sends a Line Feed (HEX 0A) after the text for the interaction. Quite a lot of devices will only respond once a Carriage Return (HEX 0D) has been sent. 

**Describe the solution you'd like**
Ability to either included "special" characters within the text for a serial interaction, or the ability to set whether 0A and/or 0D is sent.

hifiberry, 2023-05-10:
There is no maintainer for the automations today. No further development is planned on this. If you want to create a PR that's compatible with the current behaviour, we'll look into it.

chazman88, 2023-05-10:
Can you point me in the direction of the code for the interactions and ill have a look at it?

hifiberry, 2023-05-10:
https://github.com/hifiberry/create/tree/development/Beocreate2/beo-extensions/interact

hifiberry, 2023-05-10:
Just note that the file will be overwritten with every update