hifiberry/hifiberry-os #457: Wrong board detected

chazman88, 2023-05-10:
**Describe the bug**
Wrong board detected.
HifiBerryOS has detected "DAC+ Zero/Light/MiniAmp"
Board installed is "HiFi Berry Digital-Pro HW 1.2"

/boot/config.txt currently has "dtoverlay=hifiberry-dac"

according to the board it should be "dtoverlay=hifiberry-digi"

**HiFiBerryOS version**
20230404

**HiFiBerry sound card**
HiFi Berry Digital-Pro HW 1.2

[image]

[image]

hifiberry, 2023-05-10:
This board is not compatible with the Pi 4 HW 1.5. You need the latest Digi2 Pro for this Pi version.

chazman88, 2023-05-10:
Dammit!
Is there a Pi board that will work with this HifiBerry board and HifiBerryOS?

chazman88, 2023-05-10:
@hifiberry Or maybe a discount code to make having to replace my existing board a little less painful??

hifiberry, 2023-05-10:
All Pi2/3 will work fine.

closed 2023-05-17 by hifiberry