hifiberry/hifiberry-os #459: Adding the rsync package to HiFiBerry

WarnerV, 2023-05-15:
**Is your feature request related to a problem? Please describe.**
Maintaining music to a drive that is directly attached to the HiFiBerry machine is a bit of a hassle. Adding rsync as a package would greatly simplify keeping the music collection in sync without the need for network drives to be always online for all HiFiBerry instances.

**Describe the solution you'd like**
Adding the rsync package to HiFiBerry by default, so users can sync their music files in a safe, secure and dependable way.

**Describe alternatives you've considered**
1. Putting all content on a network drive could work as an alternative, but it requires an additional machine to be 'always on' (and using power), which I would love to avoid.
2. Adding Samba to HiFiBerry could also work but is more complex as a solution (plus it doesn't really replace the benefits of rsync, where you can just sync the changed files only).
3. Adding rsync through a docker container (since I have no idea about docker containers I can't evaluate the viablity at this time). Also, I think that for the general flexibility of HiFiBerry as a solution, rsync can be a nice addition: It would become possible to automatically / periodically sync content over all HiFiBerry machines with little effort.

snan, 2023-11-26:
I agree. For now, one workaround might be to use scp to put songs on there, right?  
(Still waiting on mine so I can't test yet.)

hifiberry, 2024-01-10:
We'll add it to the upcoming HBOS64

closed 2024-01-10 by hifiberry