hifiberry/hifiberry-os #465: Pull request not seen.

m1m1s1ku, 2023-07-13:
https://github.com/hifiberry/create/pull/11

Told your support team that I will do it, did it a year ago.

hifiberry, 2023-07-13:
I'm sorry that I did not reply earlier. I just read 
"(Merge not needed, mostly a tutorial with some tips & tricks.)"

This PR changes more than 200 files. That's not something we will merge. It's impossible to find out what has been changed with this amount of changes and it's impossible for us to see what might break here. 
If you want to improve the existing code, please do it in very small chunks that can be reviewed. You probable need to split this into at least 20 smaller PRs, each focussing on a specific, very small change.

m1m1s1ku, 2023-07-13:
So you don't even read what people do for you ? Lmao.
I will remember sir. And you will pay.

closed 2023-07-13 by hifiberry