hifiberry/hifiberry-os #473: Incorrect processing of WAV files

micetecumseh, 2023-08-07:
**Describe the bug**
Hello;
HiFiBerryOS's DLNA/UPnP module, working with Digi+ Pro, incorrectly processes the received 24bit **WAV** files - If you previously played a 16 bit file then it reports them to the DAC as 16bit files. 
Is the problem known and work is underway to resolve it?

Best Regards
Mice

**HiFiBerryOS version**
20230404

**HiFiBerry sound card**
Digi+Pro

**To Reproduce**
Steps to reproduce the behaviour:
1. Turn on HiFi Berry, turn on Audirvana Studio and play 24 bit file from Audirvana Studio - ADi DAC reports them properly as 24 bit
2. Then play 16 bit file from Audirvana Studio - ADI DAC reports them properly as 16 bit
3. Then again play 24 bit file from Audirvana Studio - ADi DAC reports them **improperly ** **as 16 bit**
4. There is no such problem when using the same Audirvana Studio and ADI DAC but ARCAM CDS50 instead of HiFi Berry Digi+ Pro - so HiFiBerryOS could be a reason.

**Expected behaviour**
ADI DAC always reports 24 bit files as 24 bit, not as 16 bit.

**Screenshots**
necessary

**Browser (if applicable)**
 NA

**Additional context**
Please ask if you need anything

hifiberry, 2023-08-07:
The problem is neither known nor is somebody working with it. I can't comment on what your DAC reports. Do you see the same behavior with other players?

micetecumseh, 2023-08-07:
Hello;
Thank you very much for your quick answer.

> "I can't comment on what your DAC reports" I understand this - my Pro DAC shows exactly what is sent to them so there is nothing to comment on.
> 
> Do you see the same behavior with other players?  Most probably it has nothing to do with the problem described - since with CDS50 in between is OK but with HiFiBerryOS in between it's not OK. The Player must send the truth since after passing through the CDS50 my DAC shows the truth ...

P.S.
I'll understand if you don't have the time to deal with it, but if you have the time, I'd appreciate it if you thought about it.

Best Regards;
Mice

closed 2024-03-14 by hifiberry