hifiberry/hifiberry-os #475: Digital audio output with amp2 setup

chmod750, 2023-08-10:
I am using my hifiberry amp2 mainly to stream music from spotify.
I would like to connect my TV output to my hifiberry as well. My TV has one jack output and one digital audio output plug. The jack usage is not implemented, so what sould I do to be able to connect my TV on my audio setup ?
Thanks

hifiberry, 2023-08-10:
Please use our forum https://support.hifiberry.com/hc/en-us for general questions like this. 
However, in your case there is nothing you can do anyway. There is no digital input on the Amp2. You would need the Beocreate 4-channel amplifier for this kind of configuration.

closed 2023-08-10 by hifiberry

chmod750, 2023-08-10:
Thanks, I will open a post there for futher questions :)