hifiberry/hifiberry-os #476: New release with new shairport-sync version

kaitimmer, 2023-09-20:
DO NOT remove the blocks below, but fill these with the requested data. Incomplete feature requests will be closed.
You should remove this comment before posting the feature request.

**Is your feature request related to a problem? Please describe.**
Airplay 2 support is missing, and I have random streaming aborts from my Apple TV when using the Hifiberry as the audio output device.

**Describe the solution you'd like**
The last release is quite old, and I would love to have Airplay 2 and a newer version of shairport-sync available in hifiberry-os.

**Describe alternatives you've considered**
I tried to build the image myself but ran into a lot of issues with the buildroot setup on my Mac.

**Additional context**
Reading through this: https://github.com/hifiberry/hifiberry-os/pull/466 I would think that this is already in the `development ` branch so maybe all that is needed would be a new release version.

hifiberry, 2023-09-20:
Airport2 support is already included in the latest HiFIBerryOS release.

kaitimmer, 2023-09-20:
Ah, ok, I missed that in the older Changelogs. Still would like to see the newer shairpoint-sync version in a new release to help out with the connection issues I see.

hifiberry, 2023-09-20:
New release will be published when it's ready...

hifiberry, 2024-01-12:
Feel free to test HiFiBerryOS64 (there's no direct update possible now, you have to start with a fresh installation)

closed 2024-01-12 by hifiberry