hifiberry/hifiberry-os #478: Shairport-sync security update

grischard, 2023-10-11:
DO NOT remove the blocks below, but fill these with the requested data. Incomplete bug reports will be ignored!
You should remove this comment before posting the bug report.

**Describe the bug**
Shairport-sync 4.3 and 4.3.1 contain security updates

**HiFiBerryOS version**
20230404

**HiFiBerry sound card**
any
**To Reproduce**
Check shairport-sync version

**Expected behavior**
Latest stable

**Screenshots**
n/a

**Browser (if applicable)**
n/o

**Additional context**

https://github.com/mikebrady/shairport-sync/releases mentions security updates in 4.3 and 4.3.1.

(Looking at the commits, only 4.3 contains actual security updates, and the release notes seem to have been copy-pasted)

Would you like me to do a PR like https://github.com/hifiberry/hifiberry-os/pull/466?

hifiberry, 2023-10-12:
Ye,s please send a PR.

grischard, 2023-11-08:
@hifiberry there's now a 4.3.2, but I don't want to spam you with a PR for every shairport release if you're not planning a hifiberry release right now.

PR #479 (2023-11-22): Security: Update shairport-sync and nqptp to 4.3.1
Fixes #478
closed 2024-01-10 by hifiberry