hifiberry/hifiberry-os #480: Bluetooth remote support is broken at the kernel level

jkp, 2023-10-22:
DO NOT remove the blocks below, but fill these with the requested data. Incomplete bug reports will be ignored!
You should remove this comment before posting the bug report.

**Describe the bug**
My Bluetooth remote (the [teenage engineering OR1](https://teenage.engineering/guides/or-1)) stopped working since the `20211105` release.

**HiFiBerryOS version**
I bisected by flashing historic versions of the OS and found out that any release after `20211105` is broken.

**HiFiBerry sound card**
Post the exact type, e.g. DAC+ Pro instead of "DAC+".

**To Reproduce**
Steps to reproduce the behavior:
1. Pair the remote to the system using `bluetoothctl`
2. Run `ls /dev/input`: no device node is registered with the system.

**Expected behavior**
A device node should be registered with the system

**Screenshots**
N/A

**Browser (if applicable)**
N/A

**Additional context**
This looks to be an issue caused by a kernel update: the device pairs properly so the bluetooth stack is fully functional, however the kernel doesn't load a device driver. It broke between the `20211105` and `20220228` releases which at least gives a narrow range of commits to focus on.

hifiberry, 2024-01-10:
We don't have any BT remote controls to test this. If you can find the root cause for the problem, feel free to provide a patch.

closed 2024-01-10 by hifiberry

jkp, 2024-03-27:
Running the latest `HiFiBerryOS64` to see if this is resolved and unfrotunately its still an issue :|

I'm happy to try to find a resolution but it requires a kernel rebuild ... is there any documentation about how to build a custom  kernel for this system?

hifiberry, 2024-03-27:
HiFIBerryOS64 uses the latest official Raspberry Pi kernel. What exactly do you want to "rebuild"?

jkp, 2024-03-27:
Check out my original bug report above. In older releases (pre `20211105`) I could connect a bluetooth remote and it would register an input device which would then feed events straight to `audiocontrol2` via the keyboard module. This was broken after that release. I'm still able to pair the remote using `bluetoothctl` but a device node is no longer instantiated by the kernel.

If you're using a stock kernel that's good to know...I guess it means I can broaden my hunt to see if any similar behaviour is known in other projects that use the same kernel...

But open to ideas here: any suggestions why this would have broken?

hifiberry, 2024-03-27:
No, I don't know any details about the BT stack and the BT drivers in the kernel.

jkp, 2024-03-28:
I’ve played around with a bunch of vanilla installs on various pis now and they all have the same problem. So this is not hifiberry specific

If anyone knows where the appropriate upstream place to raise this issue would be I’ll happily chase it that way?

Thanks

Sent from [Proton Mail](https://proton.me/mail/home) for iOS

hifiberry, 2024-03-28:
I would recommend to contact the manufacturer. They should understand better than anyone else how this should work.

jkp, 2024-04-04:
OK, I won't be beaten here! I've been playing around for days trying to track the bug down and at least part of it I have figured out.

In the release I mentioned in my original post there was an upgrade to `buildroot` that in turn had the side-effect of disabling a bunch of stuff in `bluetoothd` that [was previously enabled by default](https://github.com/buildroot/buildroot/commit/fb9fc969d91f69851b4c70dba512b607e52bda7b): in this case specifically the issue is HID and HOG support (see your [buildroot config files](https://github.com/hifiberry/hifiberry-os/blob/master/configs/hifiberryos-nogui#L3285)).

I've confirmed that with the older `bluetoothd` binary copied onto a newer system the nodes in `/dev/input` are created correctly: this will directly be to the missing plugins that have been compiled out in later versions of hifiberry. To bring back parity it would just need someone to review the options that have been set in the configs and flip a few of them to true (I'd also love to see `btmon` back on the system!)

jkp, 2024-04-04:
> I’ve played around with a bunch of vanilla installs on various pis now and they all have the same problem. So this is not hifiberry specific If anyone knows where the appropriate upstream place to [...]

This turns out to have been an incorrect statement as it goes...there's a second issue downstream of the input nodes being created that seems to be present on many more systems that I'm continuing to hunt down. I'll post back as I find more...

jkp, 2024-04-04:
And the second issue...seems to be an issue on the side of the remote hardware I'm using. It seems that sometimes it gets stuck in some funky state and I need to remove the batteries and start totally from scratch for the device to function properly once paired. Once I do this everything works. So in summary, if we could get those missing plugins back into the bluetooth config things should likely work again as intended on newer builds!

hifiberry, 2024-04-05:
I don't know what "missing plugins" you're refering to. We update software regularly, it's possible that something has been removed from the standard options in later releases. Feel free to provide a pull request, but we won't dig into this further.