hifiberry/hifiberry-os #482: Change Password dialog nonfunctional

lanxique, 2023-11-16:
**Describe the bug**
The "Change Password" function does not allow for a password change after the initial default is changed.

**HiFiBerryOS version**
20230404

**HiFiBerry sound card**
Digi+

**To Reproduce**
1. Change password from default
2. Go back to General -> Remote Login
3. Click Change Password
4. In the modal "Change System Password" box, enter the current password (NB: will fail if an incorrect password is entered)

**Expected behavior**
A dialog should offer the option to enter a new password; instead, the "Change System Password" box closes and returns to the Remote Login page.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Browser (if applicable)**
 - OS: Windows 10
 - Browser: Both Edge and Firefox
 - Version 119 (of both)

**Additional context**
Remote Login via SSH is working as expected

hifiberry, 2024-01-19:
I can't reproduce this problem here. Can you please try the lastest HiFiBerryOS 64 to see if the problem is reproducible with this (it should be as there was no change in the UI code).

wisp3rwind, 2024-02-06:
Same problem here. I suspect that this is somehow related to not properly escaping the password: This was with the randomly generated password `&ghcw$CMTSxZ3q&u%PDJ` (not using that anymore). I've re-installed HifiberryOS to fix this problem, and now chosen a password with alphanumeric characters only, which seems to work.

This is on
- HiFiBerryOS64 alpha 5
- Firefox 122.0

wisp3rwind, 2024-02-06:
> Same problem here. I suspect that this is somehow related to not properly escaping the password: This was with the randomly generated password `&ghcw$CMTSxZ3q&u%PDJ` (not using that anymore). I've [...]

EDIT: Reading the original post again, it might not be the same issue: SSH login didn't work for me with that password.

hifiberry, 2024-02-06:
You're saying, changing the password works, but only with alphanumeric characters?

wisp3rwind, 2024-02-06:
To be more specific, these are the steps that I took:

- fresh Hifiberry setup
- change the password (`hifiberry` -> `&ghcw$CMTSxZ3q&u%PDJ`), this appears to work
- (I think I might have rebooted in the meantime)
- try to login via ssh: rejects both the default and the new password
- try to change the password again: when entering either of the two passwords in the prompt asking for the current password, that prompt simply closes (If I remember correctly, there was no error message; it certainly didn't advance to the next step allowing to enter a new password)

After a fresh install, I used a password containing only `0-9a-zA-Z` which seems to work (i.e. ssh login is successful). I Haven't rebooted since the first startup yet; going to try that now.

closed 2024-03-19 by hifiberry

hifiberry, 2024-03-19:
Fixed with https://github.com/hifiberry/create/commit/88c6e7b1c3d2fd5c4cfdb25a149bfbad0adc3b58

wisp3rwind, 2024-03-19:
Hi, thanks for looking into this! I'm not using HifiberryOS anymore, so can't easily test, but: It looks like `child_process.exec` always runs things through a shell and I couldn't see any sanitization in the adjacent code. Thus, the password seems to be subject to expansion by the shell (variable references, etc). Single quotes instead of the new double quotes would help to prevent variable expansion, but then there's still the possibility of single quotes in the password. What about `child_process.spawn` to avoid the shell completely?