hifiberry/hifiberry-os #487: Slow startuo due to rndg using all cpu

burkhre2, 2023-12-08:
DO NOT remove the blocks below, but fill these with the requested data. Incomplete bug reports will be ignored!
You should remove this comment before posting the bug report.

**Describe the bug**
When freshly starting hifiberryos on a raspberry4 I can login with ssh and with top I see full CPU usage by rndg during a minute or more.

Also the xecution of  rngd --list hangs on TPM step

This seems related to https://bugzilla.redhat.com/show_bug.cgi?id=1739730
Work-around: 
Replace in /etc/systemd/system/multi-user.target.wants/rngd.service the line with ExecStart by
ExecStart=/usr/sbin/rngd -f --no-tpm=1

**HiFiBerryOS version**
Buildroot 2021.11
Linux hifiberry 5.15.78-v7l #1 SMP Mon Apr 3 20:04:51 UTC 2023 armv7l GNU/Linux

**HiFiBerry sound card**
Amp2

**To Reproduce**
Steps to reproduce the behavior:
1. Shut dowen hifiberryos
2. Startup hifiberryos
3. Login via ssh with root
4. Start top

**Expected behavior**
No high cpu usage during longer time

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Browser (if applicable)**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

hifiberry, 2024-01-22:
Can you try the latest HBOS 64 Alpha release? It should be fixed in this.

burkhre2, 2024-01-27:
Hi,

the slow bootup has now gone.
Also the storage of the last volume now works instead of always starting with max volume.

But now the wireless remote control does not work any more!! The usb stick for the remote is recognised, but the system does not react on any keypress. On the previous version it has worked.

Similar symptoms as described in https://support.hifiberry.com/hc/en-us/community/posts/360015923778-HiFiBerry-Remote-Control

Regards

Rene Burkhardt

closed 2024-01-28 by hifiberry