hifiberry/hifiberry-os #490: HiFiBerryOS 64 not starting on RPI 3B+

mbolo01, 2023-12-18:
DO NOT remove the blocks below, but fill these with the requested data. Incomplete bug reports will be ignored!
You should remove this comment before posting the bug report.

**Describe the bug**
HiFiBerryOS 64 does not start on my RPI 3B+

**HiFiBerryOS version**
HiFiBerryOS 64

**HiFiBerry sound card**
DAC+DSP

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
RPI to start HiFiBerryOS

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Browser (if applicable)**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Additional context**
I created a SD card from the Github zip file using balenaEtcher
I inserted the SD card into my RPI 3B+ rev 1.3 previously booting on another SD card with HiFiBerryOS 20230404
The RPI is wired to the network.
I connected an HDMI display to control startup
Nothing happens post power on (nothing on the display, no DHCP request, no LED on the RPI showing any activity.

npmstart-pray, 2023-12-18:
You've flashed the first alpha version to your pi.
Being an alpha, there will be problems and issues and bugs that need sorting. 
you may have discovered a significant one. 

But: a 64bit OS is aimed at the new pi 5, as well as the pi 4, both with 8GB of RAM. It may cause more trouble than solve problems on -your- Pi.

hifiberry, 2023-12-19:
@mbolo01 Thank you for the bug report. You're absolutely right. Unfortunately in the latest build there are some files missing :( That happens if you only test the image on a single Pi ;-)

I hope we can upload another build later today.

hifiberry, 2023-12-19:
Can you please try the new release:
https://github.com/hifiberry/hifiberry-os/releases/tag/64alpha2

@npmstart-pray As this will be the future platform for all 64-bit capable Pi's, it's important to have testers for all platform, even if the memory is <4GB.

hifiberry, 2023-12-20:
Looks like the config.txt was not edited correctly. If you want to try, you can fix it by editing /boot/config.txt

mount -o remount,rw /boot 

Then edit config.txt and put this at the end (the arm_64bit should be there already).

# enable 64bits support
arm_64bit=1
dtparam=i2c=on
dtparam=spi=on
kernel=Image
force_eeprom_read=0
dtoverlay=i2c-gpio,i2c_gpio_sda=0,i2c_gpio_scl=1

About the DSP programs. As v15 should now include all bugfixes of previous releases and be compatible with all DAC+DSP boards, we would like to get rid of the older versions in HiFiBerryOS. You can still copy these to the system, but we have seen people having trouble trying to get these working on the latest hardware. Therefore we don't want to include these in the default setup. 
It would be also good if you could test the v15.

mbolo01, 2023-12-20:
DAC + DSP is now recognized, tks.
FYI: At first start, DSP Program is listed as "unknown program", I had to install the DAC+DSP Universal v.15 which works fine with my setup where I have an Optical Input.
I'll test Airplay, Optical Input, Bluetooth, Radio and my main music source which is Plexamp running in a docker container.
I'll also run a new Room Compensation
I will not test the following sources: Music Room, Spotify, DLNA, OpenHome, Snapcast and Squeezlite.

mbolo01, 2023-12-20:
I have an issue with Plexamp running in a Docker container: While the installation was OK, the player fails to play anything. Looking at the list of audio devices from a Plexamp perspective, “snd_rpi_hifiberry_dac: HifiBerry DAC HiFi pcm5102a-hifi-0” which is usually listed is not present here. 
Looking at the Plexamp docker installation, I noticed that Plexamp runs as "plexamp" user. I checked on my running 32 bits instance and required plexamp user and group are indeed present, I must have created them at first install.
The issue with the HFBOS 64 is that useradd and groupadd commands are not present.

hifiberry, 2023-12-20:
We never had useradd/userdel in HBOS, you probably have edited /etc/passwd by hand. 

However: Do you use a 64bit version of the PlexAmp binary? 32bit binaries won't run on the 64bit OS.

mbolo01, 2023-12-20:
You are right, my fault ... I was looking for user/group in HBOS instead of inside Docker container. So all is in order in the Docker container, I have installed an ARM64 instance of Plexamp, docker image starts properly and I can access Plexamp UI and perform various actions but playing music.
I have checked Plexamp logs which are clearly showing an issue dealing with the audio interface.
Any idea why  “snd_rpi_hifiberry_dac: HifiBerry DAC HiFi pcm5102a-hifi-0”  is exposed as an audio device to my 32 bit/docker container setup but not to my 64 bit/docker container ? any command line I could run to get some info ?

mbolo01, 2023-12-20:
First my apologies, I'm not a UNIX expert, so I may not fully understand everything we talk about here.
From the docker container used to run plexamp, I ran "aplay -l" as plexamp user (the user that is used to run plexamp) and indeed there are some issue when the user is not "root". Note that I added the plexamp user in audio the container group as I found this being a potential solution for this kind of problems:

rooot@hifiberry:~# su - plexamp
plexamp@hifiberry:~$ aplay - l
ALSA lib confmisc.c:767:(parse_card) cannot find card '0'
ALSA lib conf.c:4745:(_snd_config_evaluate) function snd_func_card_driver returned error: No such file or directory
ALSA lib confmisc.c:392:(snd_func_concat) error evaluating strings
ALSA lib conf.c:4745:(_snd_config_evaluate) function snd_func_concat returned error: No such file or directory
ALSA lib confmisc.c:1246:(snd_func_refer) error evaluating name
ALSA lib conf.c:4745:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5233:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2660:(snd_pcm_open_noupdate) Unknown PCM default
aplay: main:830: audio open error: No such file or directory

Maybe some interesting information above for you ?

hifiberry, 2023-12-20:
Did you add it to the audio group in the container or on the "main" system? I don't think doing this in the container will work (never tried it). You probably need to create the user outside the container and add it to the audio group outside the container. I haven't tested this yet, so just an idea here. I might find some time to look into this tomorrow, but can't guarantee it.

mbolo01, 2023-12-20:
I added the plexamp user manually in the pass file as well as in group file for the audio group, no change. Over to you, whenever you can.

closed 2024-01-10 by hifiberry

off112a, 2024-03-31:
hi @mbolo01 I have the same problem. How did you change the container Audio GID to 2001?

off112a, 2024-03-31:
@mbolo01 thank you. It worked 😀