hifiberry/hifiberry-os #491: HifiBerryOS 64 - Optical In / Auto-Stop other sources always ON

mbolo01, 2023-12-21:
DO NOT remove the blocks below, but fill these with the requested data. Incomplete bug reports will be ignored!
You should remove this comment before posting the bug report.

**Describe the bug**
When Optical In Auto-Stop other sources is turned off, it stills behaves like if it was ON

**HiFiBerryOS version**
You find this on the "About" screen. It looks like Software 20YYMMDD

**HiFiBerry sound card**
DAC + DSP

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
I'm expecting Optical In not to Auto-stop others sources when playing.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Browser (if applicable)**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

hifiberry, 2024-01-19:
I just check this. While TOSLink mutes other sources, it does not stop them. With the DSP profile we supply, TOSLink input always takes precedence over sources on the Pi.
So, if you don't want to have your sources interrupted by TOSLink, you have to disable it.

mbolo01, 2024-01-19:
So this a change from previous profile that unfortunately does not suit my setup anymore.
My TV is TOSLink connected to the DAC+DSP and I like being able to watch TV while listening to another source. With current profile TV Sound is not muting the other sources unless I paused them.  With your new one it is no longer the case. I'll have to find a other solution for that then.

closed 2024-01-22 by hifiberry