hifiberry/hifiberry-os #492: HifiBerryOS 64 - Could not find a way to apply Room Compensation results

mbolo01, 2023-12-21:
DO NOT remove the blocks below, but fill these with the requested data. Incomplete bug reports will be ignored!
You should remove this comment before posting the bug report.

**Describe the bug**
After running Room Compensation, I din't find a way to apply the saved results in the Sound Design

**HiFiBerryOS version**
20231219

**HiFiBerry sound card**
DAC+DSP

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
Being able to apply the results of the Room Compensation analysis in the Sound Design

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Browser (if applicable)**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

hifiberry, 2024-01-19:
You applay the room equalization directly in the "Room equalisation" screen. Just check, it's working fine.

closed 2024-01-19 by hifiberry

hifiberry, 2024-01-19:
Switching listening modes also works here. Please attach a full way how to reproduce the issue with screen shots. 

You just have to click on the Room EQ in the Room equalisation screen to apply it. There is no "Apply" button.

mbolo01, 2024-01-19:
This is what I have in the Room Compensation screen and when I click on the one I have recorded. Does it mean that the only one I have is selected by default ?

I'll send screenshots for the listening mode when I can get the control on the PI back, not sure why it is requesting two DHCP adresses while I think I disabled the WiFi in it.

hifiberry, 2024-01-19:
I don't think replying by email with attachments does work. You should reply directly in Github.

mbolo01, 2024-01-19:
[image]
[image]
Here are the Room Compensation screen I have:

hifiberry, 2024-01-19:
You have to click on "New room compensation" to calculate the compensation.

mbolo01, 2024-01-19:
Okay ... my fault. Was confused by the "New" term and I didn't remember the only one I did months ago. Sorry for that

hifiberry, 2024-01-19:
Ok, we might change the naming of the button.

mbolo01, 2024-01-19:
Also, I can use a saved sound design too. I think I was confused too by the screen sequence, clear to me now.