hifiberry/hifiberry-os #493: HifiBerryOS 64 - DAC HD Plus is not recognized

diytim, 2023-12-27:
DO NOT remove the blocks below, but fill these with the requested data. Incomplete bug reports will be ignored!
You should remove this comment before posting the bug report.

**Describe the bug**
Hardware: Raspberry Pi 4, 2GB
Display connected to HDMI
16GB SD-Card
After booting hifiberry-os 64 logo is displayed, system boots several times, displays login prompt,
displays logo again. HiFiBerry GUI shows **DAC+ Zero/Light/MiniAmp** as installed soundcard.
I edited config.txt at the end to this:
\# enable 64bits support
arm_64bit=1
dtparam=i2c=on
dtparam=spi=on
kernel=Image
force_eeprom_read=0
dtoverlay=i2c-gpio,i2c_gpio_sda=0,i2c_gpio_scl=1
dtoverlay=hifiberry-dacplushd

I can start playing, see album cover but no sound is heard.

**HiFiBerryOS version**
20231219

**HiFiBerry sound card**
DACPLUSHD

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
Sound should be played with DACPLUSHD card.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Browser (if applicable)**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

diytim, 2023-12-27:
Forgot: hifibery os 64 version: alpha2

hifiberry, 2024-01-19:
This should be fixed in the latest alpha. Can you check?

diytim, 2024-01-19:
Thanks a lot.
Works now as expected.

closed 2024-01-22 by hifiberry