hifiberry/hifiberry-os #495: Bluetooth: after unpairing a device, I cannot pair it again

pmts, 2023-12-29:
**Describe the bug**
The first time I connect my phone to Hifiberry via bluetooth, it works correctly. However, if I remove the pair on my phone and try to pair it again, this results in the following error: "Failed to connect to HiFiBerry. The PIN or pair code is incorrect". The only way I found to restore that phone's ability to connect to the Hifiberry is to ssh to it and use the `bluetoothctl` command to manually remove the device:

```
# bluetoothctl
Agent registered
[bluetooth]# remove C8:9B:D7:08:0F:6C
[DEL] Device C8:9B:D7:08:0F:6C realme 9 Pro+
Device has been removed
[bluetooth]#
```

After this, I can pair my device again, but only once.

I reproduced this behaviour on 2 different android phones (Realme and Xiaomi) and on Windows 10, so it's not device dependent. 

**HiFiBerryOS version**
20230404

**HiFiBerry sound card**
DAC + DSP

**To Reproduce**
1. Enable bluetooth and start pairing on Hifiberry
2. Pair a device
3. Unpair the device
4. Try to pair the same device again
5. The pairing fails

**Expected behavior**
The device pairs succesfully

**Browser (if applicable)**

**Additional context**

jonathanota, 2024-01-04:
I have experienced a similar issue with the 20230404 release on a Raspberry Pi4 with DAC2 Pro. 

In addition, when attempting to connect via my Intel Macbook Pro laptop, it asks for a pin code.

Weidav, 2024-03-18:
I have the same issue with my Raspberry Pi4, I'm pretty sure this problem exists on every device and this needs to be addressed.
Until then there is a temporary solution:

1. connect to your Raspberry Pi via ssh
2. find all paired bluetooth devices with `bluetoothctl paired-devices` and copy the MAC-Adress of the device you are trying to reconnect.
3. remove this device with `bluetoothctl remove XX:XX:XX:XX:XX:XX` (replace the X's with your actual MAC-Adress)
4. restart the bluetooth service `service bluetooth restart` (it shouldn't be necessary but won't hurt)

hifiberry, 2024-03-19:
@Weidav  Thank you for your workaround. 

While we will be looking into some way to address this, this is not very high on our priority list. If somebody wants to look into this, we're happy to integrate contributions.

Weidav, 2024-03-21:
I'd like to look into this and see how far I can get. The contribution guidelines [link here](https://github.com/hifiberry/hifiberry-os/blob/master/doc/contributions.md) isn't working. Are there any guidelines I need to have a look at?

hifiberry, 2024-03-21:
We're talking about a contribution to the GUI code - right?

In this case, base it on the "development" branch of https://github.com/hifiberry/create

Weidav, 2024-03-21:
Yes, I think a component that displays the content from my workaround should do the job.

RavanH, 2024-07-08:
The discussion on https://support.hifiberry.com/hc/en-us/community/posts/4410009712657-Cannot-connect-Bluetooth-to-hifiberryOS sounds related...