hifiberry/hifiberry-os #496: References to spotifyd in documentation - should this be vollibrespot instead?

niklasdewally, 2023-12-31:
**Is your feature request related to a problem? Please describe.**

I was trying to see what further configuration of the spotify daemon I could do within hifiberry.

According to the docs Spotify uses `spotifyd` and the associated `spotifyd.conf` (https://github.com/hifiberry/hifiberry-os/blob/master/doc/services.md).

However, the `/etc/spotifyd.conf` file is empty, and both `systemd` and the webui indicates that HifiBerry uses `vollibrespot`.

There is a `vollibrespot.conf` file which I presume is what I need to change, but this was not indicated in the docs.

**Describe the solution you'd like**

The documentation should be updated to reflect the current services and config files used (`vollibrespot` and `vollibrespot.conf`).

I am not sure if `/etc/spotifyd.conf` is present on a clean-build of the current `main` or not, but if so it should be removed entirely.

**Describe alternatives you've considered**
N/A

**Additional context**
Hifiberry version

hifiberry, 2024-01-15:
Yes, we're using Vollibrepot today. Feel free to directly edit the documention with the "Edit this file" functionality of Github and send a PR.

hifiberry, 2024-01-15:
volllibrespot doesn't use dbus anymore. We have replaced spotifyd by vollibrespot as the dbus interface on spotifyd often stopped working :(

With more containarization in the future we hope to integrate players easier. Maybe that's another chance for spotifyd. But it's not in the a short term planning.

closed 2024-01-15 by hifiberry