hifiberry/hifiberry-os #497: HifiBerry Os 64 alpha2: sigmatcpserver and store-volume failing

innir, 2024-01-02:
You might already know, but I thought I'd better report it anyway

**Describe the bug**
sigmatcp.service and store-volume.service fail to start

**HiFiBerryOS version**
HifiBerry Os 64 alpha2

**HiFiBerry sound card**
DAC+ DSP on Raspberry Pi Zero 2w

**To Reproduce**
journalctl -e

```
Jan 02 20:30:50 hifiberry (cpserver)[482935]: sigmatcp.service: Referenced but unset environment variable evaluates to an empty string: SIGMATCPSERVER_ARGS
Jan 02 20:30:53 hifiberry sigmatcpserver[482935]: Traceback (most recent call last):
Jan 02 20:30:53 hifiberry sigmatcpserver[482935]:   File "/usr/bin/sigmatcpserver", line 36, in <module>
Jan 02 20:30:53 hifiberry sigmatcpserver[482935]:     import hifiberrydsp.server.sigmatcp
Jan 02 20:30:53 hifiberry sigmatcpserver[482935]:   File "/usr/lib/python3.11/site-packages/hifiberrydsp/server/sigmatcp.py", line 39, in <module>
Jan 02 20:30:53 hifiberry sigmatcpserver[482935]:   File "/usr/lib/python3.11/site-packages/hifiberrydsp/hardware/adau145x.py", line 27, in <module>
Jan 02 20:30:53 hifiberry sigmatcpserver[482935]:   File "/usr/lib/python3.11/site-packages/hifiberrydsp/hardware/spi.py", line 41, in <module>
Jan 02 20:30:53 hifiberry sigmatcpserver[482935]:   File "/usr/lib/python3.11/site-packages/hifiberrydsp/hardware/spi.py", line 51, in SpiHandler
Jan 02 20:30:53 hifiberry sigmatcpserver[482935]:   File "/usr/lib/python3.11/site-packages/hifiberrydsp/hardware/spi.py", line 30, in init_spi
Jan 02 20:30:53 hifiberry sigmatcpserver[482935]: FileNotFoundError: [Errno 2] No such file or directory
Jan 02 20:30:53 hifiberry systemd[1]: sigmatcp.service: Main process exited, code=exited, status=1/FAILURE
Jan 02 20:30:53 hifiberry systemd[1]: sigmatcp.service: Failed with result 'exit-code'.
Jan 02 20:30:53 hifiberry systemd[1]: Failed to start SigmaTCP Server for HiFiBerry DSP.
Jan 02 20:30:53 hifiberry systemd[1]: sigmatcp.service: Consumed 2.622s CPU time.
Jan 02 20:30:57 hifiberry systemd[1]: Starting Store volume...
Jan 02 20:30:57 hifiberry systemd[1]: Stopped SigmaTCP Server for HiFiBerry DSP.
Jan 02 20:30:57 hifiberry systemd[1]: sigmatcp.service: Consumed 2.622s CPU time.
Jan 02 20:30:58 hifiberry store-volume[482954]: Traceback (most recent call last):
Jan 02 20:30:58 hifiberry store-volume[482954]:   File "/usr/bin/dsptoolkit", line 25, in <module>
Jan 02 20:30:58 hifiberry store-volume[482954]:     import hifiberrydsp.dsptoolkit
Jan 02 20:30:58 hifiberry store-volume[482954]:   File "/usr/lib/python3.11/site-packages/hifiberrydsp/dsptoolkit.py", line 41, in <module>
Jan 02 20:30:58 hifiberry store-volume[482954]:   File "/usr/lib/python3.11/site-packages/hifiberrydsp/hardware/adau145x.py", line 27, in <module>
Jan 02 20:30:58 hifiberry store-volume[482954]:   File "/usr/lib/python3.11/site-packages/hifiberrydsp/hardware/spi.py", line 41, in <module>
Jan 02 20:30:58 hifiberry store-volume[482954]:   File "/usr/lib/python3.11/site-packages/hifiberrydsp/hardware/spi.py", line 51, in SpiHandler
Jan 02 20:30:58 hifiberry store-volume[482954]:   File "/usr/lib/python3.11/site-packages/hifiberrydsp/hardware/spi.py", line 30, in init_spi
Jan 02 20:30:58 hifiberry store-volume[482954]: FileNotFoundError: [Errno 2] No such file or directory
Jan 02 20:30:58 hifiberry store-volume[482958]: cat: /etc/dspvolume.store: No such file or directory
Jan 02 20:30:58 hifiberry systemd[1]: store-volume.service: Deactivated successfully.
Jan 02 20:30:58 hifiberry systemd[1]: Finished Store volume.
Jan 02 20:30:58 hifiberry systemd[1]: store-volume.service: Consumed 1.310s CPU time.

```

**Expected behavior**
store-volume and sigmaTCP Server work as expected.

**Additional context**
Add any other context about the problem here.

hifiberry, 2024-01-10:
Looks like some SPI config might be missing. How does you /boot/config.txt looks like?

innir, 2024-01-10:
Might be, it's the default one, I use:

```
# Please note that this is only a sample, we recommend you to change it to fit
# your needs.
# You should override this file using BR2_PACKAGE_RPI_FIRMWARE_CONFIG_FILE.
# See http://buildroot.org/manual.html#rootfs-custom
# and http://elinux.org/RPiconfig for a description of config.txt syntax

start_file=start4.elf
fixup_file=fixup4.dat

kernel=Image

# To use an external initramfs file
#initramfs rootfs.cpio.gz

# Disable overscan assuming the display supports displaying the full resolution
# If the text shown on the screen disappears off the edge, comment this out
disable_overscan=1

# How much memory in MB to assign to the GPU on Pi models having
# 256, 512 or 1024 MB total memory
gpu_mem_256=100
gpu_mem_512=100
gpu_mem_1024=100

# fixes rpi (3B, 3B+, 3A+, 4B and Zero W) ttyAMA0 serial console

# enable autoprobing of Bluetooth driver without need of hciattach/btattach
dtparam=krnbt=on

# enable 64bits support
arm_64bit=1
dtparam=i2c1=on
dtoverlay=hifiberry-dac

```

hifiberry, 2024-01-11:
Ok, we'll fix this in the next release. For now, you can just add

dtparam=spi=on

to /boot/config.txt

hifiberry, 2024-01-19:
This is fixed in the latest alpha release.

closed 2024-01-19 by hifiberry