hifiberry/hifiberry-os #499: DAC2 HD + DSP incorrectly recognized as DAC+ DSP

screamnAbdab, 2024-01-06:
**Describe the bug**
Testing HiFiBerryOS 64 20231219, DAC2  HD + DSP card is incorrectly recognized as DAC+ DSP. 

**HiFiBerryOS version**
HiFiBerryOS 64 20231219

**HiFiBerry sound card**
DAC2 HD + DSP

**To Reproduce**
Fresh install of OS

**Expected behavior**
Correct identification of sound card. I believe the incorrect identification is causing additional issues but once I get this settled, if issues remain, I will create additional bug tickets.

screamnAbdab, 2024-01-06:
Perhaps this isn't a bug? I see that for DAC2HD, the recommended  dtoverly value is: dtoverlay=hifiberry-dacplushd.
But, I know that previously, in UI, the sound card was listed as DAC2HD + DSP, while it now shows as DAC+ DSP. Attempting to install the DAC program fails so my initial thought is that the card is not being recognized correctly.

screamnAbdab, 2024-01-06:
Setting dtoverlay-hifiberry-dacplush made no difference. Card still showing as DAC+ DSP in UI.

hifiberry, 2024-01-10:
Duplicate of https://github.com/hifiberry/hifiberry-os/issues/493

closed 2024-01-10 by hifiberry