hifiberry/hifiberry-os #502: Document workarounds for DAC+ ADC

sploders101, 2024-01-11:
**Is your feature request related to a problem? Please describe.**
I recently set up HifiBerryOS on a raspberry pi 4 and faced some issues getting it to work with my HB DAC+ ADC. HBOS was detecting my DAC+ ADC as a DAC+ only. When I added the dtoverlay printed on the silk screen of the HAT, it was reset. I looked in the overlay folders and found it had been renamed from `hifiberry-dacadc` to `hifiberry-dacplusadc` (this was indeed the plus model that had the old overlay written on it). Simple enough, but it still didn't work. Every time the device would boot, it would reset back to the `hifiberry-dacplus` overlay. I did quite a bit of digging which led me here:

https://support.hifiberry.com/hc/en-us/community/posts/360015033257-DAC-ADC-no-input-but-output-works-and-I-see-capture-device-at-100-in-alsamixer-

This didn't initially seem to address my problem exactly, or offer a clear solution, but I used the config.txt snippet the OP provided (which he claimed didn't work), and it worked. The important line appears to be `force_eeprom_read=0`. Judging by the name, I'd imagine this automatically disables auto-detection of the HAT and prevents this auto-revert behavior, which it sounds like happens because the DAC+ and DAC+ ADC use the same EEPROM data.

For archival purposes, the configuration needs both lines:

```
[all]
dtoverlay=hifiberry-dacplusadc
force_eeprom_read=0
```

**Describe the solution you'd like**
I would like to see this workaround documented in the OS install guide, wiki, or readme. Especially since analog input is specifically listed as a feature for the OS, there should be clear instructions on how to enable it. It was easy enough for me to figure it out, but many other individuals might not be so lucky to try something someone claimed didn't work. Even just a simple note along the lines of:

> For HifiBerry DAC+ ADC boards, make sure to add the following lines to the end of `config.txt` before inserting the SD card into the Pi:
> ```
> dtoverlay=hifiberry-dacplusadc
> force_eeprom_read=0
> ```

This would provide a clear indication to owners of this device as to what's wrong and why the OS just reverts the changes they correctly make.

**Describe alternatives you've considered**
I'm not sure alternatives are necessary here due to the simple, non-functional nature of the request.

**Additional context**
I would also be happy to make this change myself if someone could direct me where it should go.

sploders101, 2024-01-12:
After using the system for a while, I found it was still overwriting my configuration after a reboot. I did some digging and found these files:

/opt/hifiberry/bin/detect-hifiberry: https://github.com/hifiberry/hifiberry-os/blob/master/buildroot/package/hifiberry-tools/detect-hifiberry
/usr/lib/systemd/system/hifiberry-detect.service: https://github.com/hifiberry/hifiberry-os/blob/master/buildroot/package/hifiberry-tools/hifiberry-detect.service

It appears these files work to add a service that is loaded on boot which auto-detects which hifiberry HAT you've installed. This is quite useful, except when the failsafe is to just scrap the user's config and restart... `/opt/hifiberry/bin/readhat` is unable to read my HAT for some reason. I can see solder points for what appears is *supposed* to be an I2C EEPROM module, but there isn't one. This board came like this directly from HifiBerry, and it seems I'm not alone. Many of the earlier batches have this issue. I would recommend restarting this to make the failsafe fail *safe* rather than making assumptions about what is/isn't there and overwriting the user's perfectly valid configuration. Something like a comment: `#auto-generated by hifiberry-detect. Remove for manual edits.`, which disables the detection service when removed could fix the issue.

Anyway, long story short, the following command, when run as root, fixes this issue too.

```bash
systemctl disable hifiberry-detect
```

As before, I'd be happy to contribute, but I want to make sure a proper solution is determined before starting the work. Thanks!

hifiberry, 2024-01-13:
Have a look at the detect-hifiberry script. It doesn't overwrite the configuration if a sound card is configured and detected correctly. It only does this, if it isn't (or the hifiberry-dac overlay has been configured). So, something isn't "perfectly valid" in your setup.
readhat is also not the only way to detect a sound card (as not all our cards have a HAT configuration EEPROM).

If you can find a bug in the code, let us know.

hifiberry, 2024-04-10:
Closed due to inactivity

closed 2024-04-10 by hifiberry