hifiberry/hifiberry-os #503: Rebase on the lastest librespot library

rockynox, 2024-01-13:
**Is your feature request related to a problem? Please describe.**

I am unable to initiate playback from a macOS on Spotify. When I select my Rpi with Hifiberry from my Mac, the Vollibrespot extension crashes. (see the logs below)

[This issue is known](https://github.com/librespot-org/librespot/issues/1179) and is in the process of being resolved on the most active [librespot repo](https://github.com/librespot-org/librespot). (something linked with the authentication)
However, I noticed that Hifiberry relies on older versions of Vollibrespot and librespot from a [repo that is no longer maintained](https://github.com/ashthespy/Vollibrespot).
Is there a specific reason why the librespot-org repo was not chosen?

Thanks ! 

**Describe the solution you'd like**
To be able to initiate playback using Spotify from any device.

**Describe alternatives you've considered**
It's working if I initiate playback from an iOS first (authentication works from there) ans then control it from macOS.

**Additional context**
Here are the logs that helped me trace the issue:

```
vollibrespot[806]: [Vollibrespot] : Country: "FR"
Jan 13 15:37:07 lalabox vollibrespot[806]: [Vollibrespot] : Metadata pipe established
Jan 13 15:37:07 lalabox vollibrespot[806]: [Vollibrespot] : Event: Volume { volume_to_mixer: 32767 }
Jan 13 15:37:07 lalabox python3[440]: ERROR: vollibrespot - Don't know what to do with vollibrespot v0.2.4
Jan 13 15:37:08 lalabox vollibrespot[806]: [Vollibrespot] : Fetching autoplay context uri
Jan 13 15:37:08 lalabox vollibrespot[806]: [Vollibrespot] : Event: SessionActive { became_active_at: 1705160228022 }
Jan 13 15:37:08 lalabox vollibrespot[806]: [Vollibrespot] : SessionActive!
Jan 13 15:37:08 lalabox vollibrespot[806]: [Vollibrespot] : Event: PlaybackStarted { track_id: SpotifyId { id: 41050848205632260281109145437504365545, audio_type: Track } }
Jan 13 15:37:08 lalabox vollibrespot[806]: [Vollibrespot] : Event: PlaybackLoading { track_id: SpotifyId { id: 41050848205632260281109145437504365545, audio_type: Track } }

⬇️
Jan 13 15:37:08 lalabox vollibrespot[806]: [Vollibrespot] : error 403 for uri hm://keymaster/token/authenticated?client_id=##############&scope=streaming,user-read-playback-state,user-modify-playback-state,user-read-currently-playing,user-read-private
Jan 13 15:37:08 lalabox vollibrespot[806]: [Vollibrespot] : Error: MercuryError
⬆️

Jan 13 15:37:08 lalabox vollibrespot[806]: [Vollibrespot] : Autoplay uri resolved to <"spotify:station:playlist:37i9dQZF1DZ06evO1vjXFQ">
Jan 13 15:37:08 lalabox vollibrespot[806]: [Vollibrespot] : Resolving uri "spotify:station:playlist:37i9dQZF1DZ06evO1vjXFQ"
Jan 13 15:37:08 lalabox vollibrespot[806]: thread 'main' panicked at 'cannot poll Map twice', /home/matuschd/buildroot-4/host/share/cargo/registry/src/index.crates.io-6f17d22bba15001f/futures-0.1.31/src/future/map.rs:35:29
Jan 13 15:37:08 lalabox vollibrespot[806]: stack backtrace:
Jan 13 15:37:08 lalabox vollibrespot[806]:    0:       0x5566648ed8 - <unknown>
Jan 13 15:37:08 lalabox vollibrespot[806]:    1:       0x556666eb98 - <unknown>
Jan 13 15:37:08 lalabox vollibrespot[806]:    2:       0x5566645dd8 - <unknown>
Jan 13 15:37:08 lalabox vollibrespot[806]:    3:       0x5566648d24 - <unknown>
Jan 13 15:37:08 lalabox vollibrespot[806]:    4:       0x556664a194 - <unknown>
Jan 13 15:37:08 lalabox vollibrespot[806]:    5:       0x5566649f84 - <unknown>
Jan 13 15:37:08 lalabox vollibrespot[806]:    6:       0x556664a684 - <unknown>
Jan 13 15:37:08 lalabox vollibrespot[806]:    7:       0x556664a54c - <unknown>
Jan 13 15:37:08 lalabox vollibrespot[806]:    8:       0x55666492e4 - <unknown>
Jan 13 15:37:08 lalabox vollibrespot[806]:    9:       0x556664a2b0 - <unknown>
Jan 13 15:37:08 lalabox vollibrespot[806]:   10:       0x55661d1d70 - <unknown>
Jan 13 15:37:08 lalabox vollibrespot[806]:   11:       0x55661d1d3c - <unknown>
Jan 13 15:37:08 lalabox vollibrespot[806]:   12:       0x556635d438 - <unknown>
Jan 13 15:37:08 lalabox vollibrespot[806]:   13:       0x556626f938 - <unknown>
Jan 13 15:37:08 lalabox vollibrespot[806]:   14:       0x556622e038 - <unknown>
Jan 13 15:37:08 lalabox vollibrespot[806]:   15:       0x5566225d60 - <unknown>
Jan 13 15:37:08 lalabox vollibrespot[806]:   16:       0x55661fa370 - <unknown>
Jan 13 15:37:08 lalabox vollibrespot[806]:   17:       0x55661ee744 - <unknown>
Jan 13 15:37:08 lalabox vollibrespot[806]:   18:       0x55661f5dd4 - <unknown>
Jan 13 15:37:08 lalabox vollibrespot[806]:   19:       0x55661edb90 - <unknown>
Jan 13 15:37:08 lalabox vollibrespot[806]:   20:       0x556622f234 - <unknown>
Jan 13 15:37:08 lalabox vollibrespot[806]:   21:       0x556621311c - <unknown>
Jan 13 15:37:08 lalabox vollibrespot[806]:   22:       0x5566213660 - <unknown>
Jan 13 15:37:08 lalabox vollibrespot[806]:   23:       0x556663f664 - <unknown>
Jan 13 15:37:08 lalabox vollibrespot[806]:   24:       0x556622f89c - <unknown>
Jan 13 15:37:08 lalabox vollibrespot[806]:   25:       0x7fbd5d73a0 - <unknown>
Jan 13 15:37:08 lalabox vollibrespot[806]:   26:       0x7fbd5d7498 - __libc_start_main
Jan 13 15:37:08 lalabox vollibrespot[806]:   27:       0x55661d2330 - <unknown>
Jan 13 15:37:08 lalabox vollibrespot[806]:   28:                0x0 - <unknown>
Jan 13 15:37:08 lalabox vollibrespot[806]: [Vollibrespot] : EventSender disconnected
Jan 13 15:37:08 lalabox systemd[1]: spotify.service: Main process exited, code=exited, status=101/n/a
Jan 13 15:37:08 lalabox systemd[1]: spotify.service: Failed with result 'exit-code'.
Jan 13 15:37:08 lalabox node[831]: Volume received from AudioControl: 83 %.
Jan 13 15:37:13 lalabox systemd[1]: spotify.service: Scheduled restart job, restart counter is at 1.
Jan 13 15:37:13 lalabox systemd[1]: Starting Vollibrespot...
```

rockynox, 2024-01-14:
My bad, I didn't check well, you're actually [relying on librespot-org](https://github.com/hifiberry/hifiberry-os/blob/3c65ae9894fb12141e70a622a60317902a85aa2d/buildroot/package/librespot/librespot.mk#L7C21-L7C61) I will try to build HifiberryOS with an up to date version of this repo to see if it's working fine.

hifiberry, 2024-01-15:
HiFiBerryOS uses this repo:
https://github.com/hifiberry/Vollibrespot

Feel free to create a patch to this repository. However, as far as I can see it's in sync with the vollibrespot master branch. We can't use the original repository as some changes are required for the HBOS integration.

closed 2024-01-15 by hifiberry

wisp3rwind, 2024-02-10:
I believe this needs to be priorized: I'm seeing constant crashes with some tracks, for example with this error (among others):

```
Feb 10 10:12:27 hifiberry vollibrespot[2119]: thread 'main' panicked at 'called `Result::unwrap()` on an `Err` value: SendError("...")', /home/matuschd/buildroot-4/host/share/cargo/git/checkouts/librespot-e4d7c35a023053af/1a224a3/playback
/src/player.rs:232:61
```

This error cannot happen in librespot since https://github.com/librespot-org/librespot/pull/823, merged in 2021, which handles this error case more gracefully.

In effect, this means that Hifiberry OS is basically unusable for me right now, far from the plug-and-play solution it aims to be (on each crash, the service is restarted, but playback is interrupted and Spotify takes a while to recover and reconnect, and even then I need to manually resume playback).

---

I've browsed the source code superficially, and it seems to me that `vollibrespot` primarily provides the following on top of `librespot`:

- a config parser instead of just taking commmandline arguments
- piping metadata to a socket

Correct me if I'm wrong, but I suspect what Hifiberry primarily needs is the metadata stream.

Given that `vollibrespot` as well as the `librespot` fork (building against `librespot` 0.4.2 doesn't work, just tested that) that it uses are unmaintained, it seems to me that what really should be done is to find a way to to use upstream `librespot` with as minimal an additional wrapper as possible.

So, what about:

- instead of having a `toml` config file with an additional parser, use the approach that `raspotify` takes: pass them as environment variables instead (documented here, just below the table: https://github.com/librespot-org/librespot/wiki/Options), which can be done by using an `EnvironmentFile=` entry in the service script, cf. the files
   
    - https://github.com/dtcooper/raspotify/blob/master/raspotify/lib/systemd/system/raspotify.service
    -  https://github.com/dtcooper/raspotify/blob/master/raspotify/etc/raspotify/conf

- use `librespot`s event system to obtain player status and metadata, cf. the documentation https://github.com/librespot-org/librespot/wiki/Events. This could be done by a simple script like https://github.com/librespot-org/librespot/blob/dev/contrib/event_handler_example.py that forwards the data to the audiocontrol service. This would probably require using a dev version of `librespot`, which has more events and sends more metadata. I'd argue this is hardly worse than using a very outdated (network-facing, talking to a service that is also being updated) library.

Assuming that this works reliably, this seems like a much more maintainable approach.

wisp3rwind, 2024-02-10:
Maybe, a more direct substitute would be to create a new fork of `librespot`, which is exactly one commit ahead of upstream, and replaces just the file https://github.com/librespot-org/librespot/blob/dev/src/player_event_handler.rs by the metadata piping from `vollibrespot`. This seems like it should be easy to maintain as well (this is probably what one would do when starting `vollibrespot` over today).

Could you confirm that the metadata/event output is really all that HBOS needs?

hifiberry, 2024-03-19:
spotifyd already has mpris support. We had to change to vollibrespot as the MPRIS server got stuck after 60 minutes. 
There is some experimental code of a spotifyd server running in docker that will be included in the next release. 
However, there is no frontend-integration today.

wisp3rwind, 2024-03-19:
For reference, I think this is what you are refering to: https://github.com/Spotifyd/spotifyd/issues/337

hifiberry, 2024-03-19:
Yes