hifiberry/hifiberry-os #505: Update via script/ssh to new HifiBerryOS64 build not working

hojo0590, 2024-01-24:
**Describe the bug**
When updating a Pi3 from the shell running an old release (i.e. 20230404 or a build done on development branch around June/July) by running the script `./update-pi 3 your_ip_here` it results in an error
```
e2fsck has unsupported feature(s): FEATURE_C12
```
after extracting the newest image on the target during checking the FS.

This most likely (I'm not deep enough into custom Pi imaging [yet]) this is caused by a newer version of `e2fsck` being used with the HifiBerryOS64 build (`e2fsck` 1.47), while the old build is using `e2fsck` 1.46.
The line that produces the error is https://github.com/hifiberry/hifiberry-os/blob/908989b63116c89c4912e41e9e9ff2431fba1f3a/buildroot/package/hifiberry-tools/resize-partitions#L89

**HiFiBerryOS version**
Build of development branch around June/July '23 to build of latest development branch.

**HiFiBerry sound card**
AMP2

**To Reproduce**
Steps to reproduce the behavior:
1. Install old (pre-HifiBerryOS64) image.
2. Build and update-script latest development branch (`./get-buildroot && ./build-config 3 && ./compile 3 &&  ./update-pi 3 ...`
3. See error

**Expected behavior**
I'd expect to simply be able to flash a new HifiBerryOS64 build onto an old build.
When manually downloading a HifiBerryOS64alpha release from github, flashing that and updating to a newly done build of development with the steps mentioned above everything works fine (but with HifiBerryOS64 `e2fsck` 1.47 is on the Pi already)

**Additional context**
Maybe this was just my incompetency and I did something wrong, just wanted to let you know in case you think this might also show up with official updates ran through the UI. I wasn't able to confirm wether or not that update mechanism also runs that `resize-partitions` script. You might want to close this right away in case...

hifiberry, 2024-01-25:
There is no update possible today from old HiFiBerryOS to the new HiFiBerryOS64 version. You need to start from a HiFiBerryOS64 image to use the update script (at least now)

hojo0590, 2024-01-25:
Yes that is what I experienced. Is it planned to allow updates through the web interface once a new HifiBerryOS64 exists? If not this should be covered in the documentation. Anyway, this can be closed.

hifiberry, 2024-01-25:
Not sure, if there will be a direct upgrade path as new the release uses a larger /boot partition.

closed 2024-01-25 by hifiberry