hifiberry/hifiberry-os #506: Docker image/container no longer present after HBOS software update

mbolo01, 2024-01-25:
DO NOT remove the blocks below, but fill these with the requested data. Incomplete bug reports will be ignored!
You should remove this comment before posting the bug report.

**Describe the bug**
Docker images and containers are removed post software updates

**HiFiBerryOS version**
You find this on the "About" screen. It looks like Software 20YYMMDD
20240124 (64 bits)
**HiFiBerry sound card**
Post the exact type, e.g. DAC+ Pro instead of "DAC+".
DAC+DSP
**To Reproduce**
Steps to reproduce the behavior:
Update HBOS
Check if previously installed docker image is still here
Attempt to start previously running docker image

**Expected behavior**
Based on your previous recommendations, I thought the way to preserve any particular setup between software updates was to use docker which is not really the case.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Browser (if applicable)**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

hifiberry, 2024-01-25:
How exactly did you create the container?

mbolo01, 2024-01-25:
based on a docker-compose.yaml file located in a subfolder of /data/docker

hifiberry, 2024-01-25:
Ok, I need to check. Can you post the file?

mbolo01, 2024-01-25:
# cat docker-compose.yaml
version: '3'
services:
  plexamp:
   container_name: plexamp
   privileged: true
   network_mode: "host"
   image: ghcr.io/pmdroid/plexamp-headless-docker:arm64-483
   volumes:
    - /data/docker/plexamp/:/home/plexamp:rw
    - /run:/run:rw
   environment:
    - RELEASE_CHANNEL=version-beta
   restart: unless-stopped

mbolo01, 2024-01-26:
"Out of the box", the plexamp extension does not work. What is the original GitHub code you relied on ? During the very first Plexamp installation there is a step that requires to input a token pulled from the Plex.tv web site. Some developers have tried to automate it, some didn't. Also, how the installation identifies that I'm running HBOS 64 ARM to pull the right code ?

mbolo01, 2024-01-27:
Update: I managed to make it work as an extension, I will update https://github.com/hifiberry/extension_plexamp with my findings

closed 2024-03-06 by hifiberry