hifiberry/hifiberry-os #507: HiFiBerryOS 64 - Alpha 4 - Spotify Connect not working

diytim, 2024-01-26:
DO NOT remove the blocks below, but fill these with the requested data. Incomplete bug reports will be ignored!
You should remove this comment before posting the bug report.

**Describe the bug**
Spotifiy Connect Plugin enabled. In Spotify HifiBerry is listed but connecting not possible 

**HiFiBerryOS version**
20240116 (OS 64 Alpha 4)

**HiFiBerry sound card**
DAC2 HD

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
Establishing connection to HiFiBerry within Spotify 

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Browser (if applicable)**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Additional context**
HiFiBerry system information:
Raspberry PI 4 Model B, Rev. 1.5
aplay 1
card 0 sndrpihifiberry
[snd_rpi_hifiberry_dacplushd]
aplay 2
device 0 HiFiBerry DAC+ HD HiFi
pcm179x-hifi-0 
[HiFiBerry DAC+ HD HiFi pcm179x-hifi-0 ]
Mixer
DAC
Features
pi localui bluetooth pi3orlater arm7
arm8 64bit
Power controller
not detected
Linux version
Linux hifiberry 6.1.72-v8 #1 SMP
PREEMPT Tue Jan 16 13:34:44 UTC
2024 aarch64 GNU/Linux
Date and time
Sat Jan 20 12:43:42 UTC 2024
 etho
192.168.17.50
/data mounted
OK
/dev/dri/card0
OK
BT devices
hci0 -
WIFI devices
wlan0
audiocontrol2
running
beocreate2
running
bluetooth
not running
bluealsa
not running
bluealsa-aplay
not running
dlnampris
not running
mpd
running
pigpio
running
raat
not running
shairport-sync
running
sigmatcp
running
snapcastempris
not running
spotify
running
squeezelite
not running
sshd
running
ympd
running
weston
running
cog
running

hifiberry, 2024-01-29:
Spotify Connect is working here just fine. However, it seems that some people are having problems with Spotify connect. The root cause is unknown.
Did this work with older versions of HiFiBerryOS?

diytim, 2024-01-29:
Hello,

thanks for answering.
Yes, with older version (not OS 64) Spotify connect worked.

Regards,

Tim Mager (diytim)

diytim, 2024-02-01:
Works now. 
Date and time were out of range. Gateway and Router ip-address wrong. Sorry for that.

hifiberry, 2024-02-01:
Thanks for the feedback.

closed 2024-02-01 by hifiberry