hifiberry/hifiberry-os #508: HiFiBerryOs 64 - Alpha 4 - HDMI Display resolution is not recognized

diytim, 2024-01-26:
DO NOT remove the blocks below, but fill these with the requested data. Incomplete bug reports will be ignored!
You should remove this comment before posting the bug report.

Describe the bug
HDMI 7" Touch Display works but full resolution is not used.
HiFiBerry GUI is shown in the right upper corner. Only about 
1/2 of screen is used.
In config.txt 'disable_overscan=1' or '# disable_overscan=1' does not 
change anything.

HiFiBerryOS version
20240116 (OS 64 Alpha 4)

HiFiBerry sound card
DAC2 HD

To Reproduce
Steps to reproduce the behavior:

    Go to '...'
    Click on '....'
    Scroll down to '....'
    See error

Expected behavior
HiFiBerry GUI should be shown in full resoltion

Screenshots
If applicable, add screenshots to help explain your problem.

Browser (if applicable)

    OS: [e.g. iOS]
    Browser [e.g. chrome, safari]
    Version [e.g. 22]

Additional context
HiFiBerry system information:
Raspberry PI 4 Model B, Rev. 1.5
aplay 1
card 0 sndrpihifiberry
[snd_rpi_hifiberry_dacplushd]
aplay 2
device 0 HiFiBerry DAC+ HD HiFi
pcm179x-hifi-0
[HiFiBerry DAC+ HD HiFi pcm179x-hifi-0 ]
Mixer
DAC
Features
pi localui bluetooth pi3orlater arm7
arm8 64bit
Power controller
not detected
Linux version
Linux[ hifiberry ](https://github.com/hifiberry)6.1.72-v8 #1 SMP
PREEMPT Tue Jan 16 13:34:44 UTC
2024 aarch64 GNU/Linux
Date and time
Sat Jan 20 12:43:42 UTC 2024
etho
192.168.17.50
/data mounted
OK
/dev/dri/card0
OK
BT devices
hci0 -
WIFI devices
wlan0
audiocontrol2
running
beocreate2
running
bluetooth
not running
bluealsa
not running
bluealsa-aplay
not running
dlnampris
not running
mpd
running
pigpio
running
raat
not running
shairport-sync
running
sigmatcp
running
snapcastempris
not running
spotify
running
squeezelite
not running
sshd
running
ympd
running
weston
running
cog
running

hifiberry, 2024-02-01:
Not really. The whole graphics stack is not really easy to compile and configure at all. This will probably one of the last things that we can fix.

diytim, 2024-02-02:
Does it make sense to test video paramters in config.txt
like described here:

 

https://elinux.org/RPiconfig#Video_mode_options

 

Regards,

Tim Mager

Display resolution is not recognized (Issue #508)

 

<https://github.com/hifiberry/hifiberry-os/issues/508#issuecomment-192164395
O3CPAVCNFSM6AAAAABCMYVP5KVHI2DSMVQWIX3LMV43OSLTON2WKQ3PNVWWK3TUHMYTSMRRGY2DG
OJVGY> .
<https://github.com/notifications/beacon/BE6OWWFFOWST5OHLVGS6ONTYRO3CPA5CNFS
M6AAAAABCMYVP5KWGG33NNVSW45C7OR4XAZNMJFZXG5LFINXW23LFNZ2KUY3PNVWWK3TUL5UWJTT

hifiberry, 2024-02-02:
Feel free to try, I can't say if this might help.

diytim, 2024-02-02:
Thanks.I´ll give you feedback.

 

<https://github.com/hifiberry/hifiberry-os/issues/508#issuecomment-192326416
SLKZAVCNFSM6AAAAABCMYVP5KVHI2DSMVQWIX3LMV43OSLTON2WKQ3PNVWWK3TUHMYTSMRTGI3DI
MJWGY> .
<https://github.com/notifications/beacon/BE6OWWHKMPRV2T3TWYK6OCDYRSLKZA5CNFS

diytim, 2024-02-10:
Here is my possible solution for configuring individual connected displays:

1. ssh to your hifiberry
2. type
   mount -o remount,rw /boot
   to make /boot/config.txt writeable
Add the following lines to /boot/config.txt
I used WINSCP (ssh with GUI for Windows Users)

```
# Userdefined HDMI mode, resolution
# hdmi_cvt=<width> <height> <framerate> <aspect> <margins> <interlace> <rb>
# width        width in pixels
# height       height in pixels
# framerate    framerate in Hz
# aspect       aspect ratio 1=4:3, 2=14:9, 3=16:9, 4=5:4, 5=16:10, 6=15:9
# margins    0=margins disabled, 1=margins enabled
# interlace    0=progressive, 1=interlaced
# rb             0=normal, 1=reduced blanking
# hdmi_group=2 means DMT (Display Monitor Timings; the standard typically used by monitors)
# hdmi_mode=87 indicates the userdefined resolution. This is the first mode after official modes (1 to 86)
# hdmi_drive=2 selects the Normal HDMI mode (with sound over hdmi cable)
# Source: http://wiki.sunfounder.cc/index.php?title=Adjust_Resolution_for_Raspberry_Pi

# example: HDMI display with resolution of 1026 x 600, ratio 16:9, margins disabled, interlace progressive, reduced blanking
hdmi_cvt=1024 600 60 3 0 0 1
hdmi_group=2
hdmi_mode=87
hdmi_drive=2
```

Reboot
Hope that helps ...

diytim, 2024-02-23:
Raspberry Pi 5 workaround:

1. ssh to your hifiberry
    type
    `mount -o remount,rw /boot`
    to make /boot/config.txt writeable

2. Copy 
    `/lib/firmware/rpi/overlays/vc4-kms-v3d-pi5.dtbo`
    to 
    `/boot/overlays/vc4-kms-v3d-pi5.dtbo`

3. Edit /boot/config.txt
    Replace at the end (depending on your previous Raspberry Pi)
    `dtoverlay=vc4-fkms-v3d-pi4,audio=off`
    with
    `dtoverlay=vc4-kms-v3d-pi5,audio=off`

pinballpower, 2024-02-24:
Thank you. This is something we can integrate in the next release.

user10072023github, 2024-09-12:
@diytim did your message here back on Feb 10th fix your issue fully when using these 64 alphas?  I followed your instructions editing config.txt according to my screen, so:
hdmi_cvt=1920 1080 60 3 0 0 0
hdmi_group=2  #dvt screen
hdmi_mode=82  #monitor is 1920x1080p
hdmi_drive=1  #dont need/want sound

I've had no luck on my end...

diytim, 2024-09-12:
Hi user10072023github,

the I needed workaround only for alpha 5 and it worked for me.
Meanwhile with latest alpha´s and RaspBerry Pi 5 it works without changing anything
in config.txt.  

Attached you´ll find HDMI troubleshooting PDF. Hope that helps.

 

Regards,

Tim

user10072023github, 2024-09-12:
@diytim Response greatly appreciated!  Sadly no attachment you mentioned came through.  I think that sadly this fix won't work for me anyway because I have a "Waveshare" 13in screen from pishop.us and not the official 7in screen.  Sigh, still stuck.

diytim, 2024-09-13:
Sorry, I didn´t know that attachments are not allowed.

Here´s the link to the troubleshooting file:

 

https://pip.raspberrypi.com/categories/685-whitepapers-app-notes/documents/RP-004341-WP/Troubleshooting-KMS-HDMI-output.pdf

 

Regards,

Tim

user10072023github, 2024-09-14:
@diytim Thank you for the link!  Unfortunately, this didn't help my issue since I have a different screen but hey I appreciate your time none-the-less