hifiberry/hifiberry-os #511: HiFiBerry OS 64 - Alpha 5 - selecting album in local library restarts connecting

diytim, 2024-02-01:
DO NOT remove the blocks below, but fill these with the requested data. Incomplete bug reports will be ignored!
You should remove this comment before posting the bug report.

**Describe the bug**
Tapping on an album in local library (USB-Stick) restarts "Connecting ...". This needs about 10 seconds.
Afterwards content of album is listed and music can be played.

**HiFiBerryOS version**
20240116 (OS 64 Alpha 5)

**HiFiBerry sound card**
HiFiBerry DAC+ HD

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
As in previous 32bit OS versions tapping on an album in local library content of
album should be listed immediately.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Browser (if applicable)**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Additional context**
HiFiBerryOS version
20240116 (OS 64 Alpha 5)

HiFiBerry sound card
DAC2 HD

HiFiBerry system information:
Raspberry PI 4 Model B, Rev. 1.5
aplay 1
card 0 sndrpihifiberry
[snd_rpi_hifiberry_dacplushd]
aplay 2
device 0 HiFiBerry DAC+ HD HiFi
pcm179x-hifi-0
[HiFiBerry DAC+ HD HiFi pcm179x-hifi-0 ]
Mixer
DAC
Features
pi localui bluetooth pi3orlater arm7
arm8 64bit
Power controller
not detected
Linux version
Linux hifiberry 6.1.72-v8 #1 SMP
PREEMPT Tue Jan 16 13:34:44 UTC
2024 aarch64 GNU/Linux
Date and time
Sat Jan 20 12:43:42 UTC 2024
etho
192.168.17.50
/data mounted
OK
/dev/dri/card0
OK
BT devices
hci0 -
WIFI devices
wlan0
audiocontrol2
running
beocreate2
running
bluetooth
not running
bluealsa
not running
bluealsa-aplay
not running
dlnampris
not running
mpd
running
pigpio
running
raat
not running
shairport-sync
running
sigmatcp
running
snapcastempris
not running
spotify
running
squeezelite
not running
sshd
running
ympd
running
weston
running
cog
running

diytim, 2024-02-01:
Raspberry Pi is connected via ethernet cable to network.

hifiberry, 2024-02-01:
I can't see this happening here. Try check the logs when you do this action.
Run the following from command line while you reproduce it:

journalctl -f

Maybe we can see something strange in the log.

diytim, 2024-02-01:
Works now. correct.
# timedatectl set-timezone Europe/Berlin
and setting correct router/gateway ip-address helped.

closed 2024-02-01 by hifiberry