hifiberry/hifiberry-os #512: DAC+ DSP EQ not working on analog outputs

jdyanine, 2024-02-02:
I have successfully implemented a Parametric EQ for room correction over the SPDIF output using the "Sound/Parametric Equalizer/Speaker Equalizer" module, and it works great. However, I encountered an issue where the same module, along with "Sound/Channels," is malfunctioning and incapable of adjusting the analog outputs. Although the features are present, they do not function over the analog outputs. It's as if the analog outputs were a pass-through signal.

However, the Sound Design module does modify the analog output. Because it applies filters globally, it is not suitable for equalizing analog outputs separately. Therefore, it does not serve as the subwoofer crossover I am looking for.

I'm using HiFiBerryOS 64 version 20240115, along with the DAC+ DSP hat. Additionally, the Update module is not functional, and I am unable to update to version 20240124 either through the webUI or SSH. I have no clue on how to do it manually.

[image]

jdyanine, 2024-02-03:
Since I couldn't update, I flashed the sdcard again to Hifimeberry OS 64 v20240124. The issue of being unable to equalize the analog outputs persists: Outputs A and B are optical, while C and D are analog, and it's not possible to modify signals C and D.

hifiberry, 2024-03-06:
I can't reproduce this here. Did you install the latest DAC+DSP Profile v15?

jdyanine, 2024-03-06:
Thank you for your reply. Yes, I'm using v15.

jdyanine, 2024-03-06:
Even if I mute the C and D channels, the analog outputs still play without
any change. I have no control over the analog outputs, only the volume and
the global EQ inside 'Sound Design.'

Thank you!

hifiberry, 2024-03-06:
I can't reproduce it here. Try re-installing the DSP profile

jdyanine, 2024-03-18:
I didn't notice any change after updating and reinstalling the DSP profile.

I've come to realize that the HifiBerry DAC+ DSP only has two channels,
rendering Channels C and D useless. I'm only able to control Channels A and
B, which affect both the optical and RCA outputs.

Therefore, it's not possible to use it as a subwoofer crossover for a 2.1
setup.

hifiberry, 2024-03-19:
That's the expected behavior. To change this, you need to create your own DSP profile.

closed 2024-03-19 by hifiberry