hifiberry/hifiberry-os #513: DAC+DSP optical input only works with Spotify Connect at same time

FatToni, 2024-02-06:
**Describe the bug**
The optical input signal is detected and can be recogniced at the UI with "Playing" indicator but will not output sound independetly.
The optical Toslink input is only working if i start/let run some musik with Spotify Connect. 
As soon I pause/stop the Spotify Connect musik the optical input sound will stop playing. 
I can only use the optical input if the spotify connect is playing some music in the background.

**HiFiBerry Version 20230404**
I tryed different DPS Versions from 11 to 15. Currently Version 15 is working with this Spotify Connect Bug.

**HiFiBerry DAC+DSP**
with Rapberry pi3 

**To Reproduce**
Steps to reproduce the behavior:
1. Start Spotify @ iphone and play musik with Spotify Connect 
2. Start optical input signal
3. Sound output will switch from Spotify to optical source
4. Sound from spotify is not hearable
5. Pause / stop spotify music, the optical input sound will also stop after a few seconds.  

**Expected behavior**
Fix the bug to use optical input independently
Thanks for your effort

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Browser (if applicable)**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.
[image]
[image]

hifiberry, 2024-02-06:
Do I understand you correctly: v15 is working?

FatToni, 2024-02-06:
V15 is the only version i get music with the optical input, but I need to use the spotify play trick in the backround. If spotify is stopped the optical input sound will also stop.

FatToni, 2024-02-09:
Problem Solved. I reset everything with clear OS installation and DSP Version 15. Now optical is working as it should.

closed 2024-02-09 by FatToni