hifiberry/hifiberry-os #515: Web App on iOS 17.4

moritzmhmk, 2024-02-17:
**Describe the bug**
Web Apps will not work with iOS 17.4 - please consider filling the form provided by https://open-web-advocacy.org/blog/its-official-apple-kills-web-apps-in-the-eu/ as a company effected by the change.

**HiFiBerryOS version**
Any

**HiFiBerry sound card**
Any

**To Reproduce**
Try to add Web App to Homescreen in iOS 17.4

**Expected behavior**
App added to Homescreen working as a standalone Web App.

**Screenshots**
Not applicable.

hifiberry, 2024-02-19:
While we can fill out this, there's nothing else we can do about. As I'm not based in the EU, I can't even test this.

closed 2024-02-19 by hifiberry

sebastianrothe, 2024-04-11:
Apple returns to being reasonable: https://open-web-advocacy.org/blog/apple-backs-off-killing-web-apps/