hifiberry/hifiberry-os #517: HiFiBerryOs 64 - Alpha - Not able to connect to WiFi on Zero 2

hanneshier, 2024-02-21:
**Describe the bug**
Im not able to connect my Pi Zero W 2 to the WiFi. The Pi does not create an Access Point, adding a wpa_supplicant.conf did not work (manually or with the Pi Imgager Tool). I was not able to connect via a display, too. When connected to a display, mouse and keyboard, the command line always changed to the Hifiberry GUI after a few seconds. There, I was not able to make any input with the mouse or keyboard.

**HiFiBerryOS version**
Tested with the following versions:
HiFiBerryOs 64 - Alpha 5
HiFiBerryOs 64 - Alpha 4
HiFiBerryOs 64 - Alpha 2

**HiFiBerry sound card**
DAC+ Zero 1.0.1

**To Reproduce**
Steps to reproduce the behavior:
1. Flash the beta to a SD card
2. (OPTIONAL: add wpa_supplicant.conf)
3. Check for Access Point
4. Connect Pi to a display

**Expected behavior**
Pi should crate Access Point or use wpa_supplicant.conf

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Browser (if applicable)**
Not applicable

**Additional context**
Not applicable

lzulauf, 2024-05-26:
I ran into this issue with Beta8 as well. I'm running a DAC2 ADC Pro on my Pi Zero 2 W. I haven't tried configuring wpa_supplicant.conf manually.

knfrmity, 2024-06-06:
I have this issue too. This isn't really my realm of expertise but I thought I would note what I found.

After some investigating, it appears that the `wlan0` device is missing entirely, so there's no way the OS can connect to existing WiFi or create the setup network. WiFi does work on a Raspberry Pi 4 running Alpha 8.

After looking through a few generic wifi on raspberry pi guides, I tried creating `/etc/network/interfaces`, as only a backup file existed, but that hasn't changed anything in terms of functionality.

jplegat, 2024-07-18:
I flashed Alpha 8 to a microSD card and tried to complete the installation on a Raspberry Pi 4 but it seems the wifi interface doesn't get activated so there's no way to connect to the ad-hoc access point and start the setup process. Just to confirm, is this a known issue?

Cascadier1, 2025-05-22:
found this post because i am unable to start a Raspberry Zero 2W after flashing the image. It stucks on the coloured screen.
What Image have you used?

Tried upgrading to HiFiBerry OS64