hifiberry/hifiberry-os #520: "Pop" from speakers attached to Amp2 at the start and end of any playback

baf, 2024-03-08:
**Describe the bug**
I am using a Pi 3B and HiFiBerry Amp2. A "pop" is heard from the attached speakers (Paradigm Micro Monitor v2) at the start and end of playback using HiFiBerry OS. This occurs when playing using LMS/Squeezelite (and any source within, e.g. Tidal, BBC Sounds, MP3 radio stream) as well as AirPlay directly to the HiFiBerry OS device. 

Using the same hardware, same power supply, and same LMS and AirPlay tests with piCorePlayer v8.2.0 does not produce the "pop" - for this test, I simply swapped the microSD card with HiFiBerry OS for one with piCorePlayer; all other variables were the same. I also have several other Pi 3 and 4 devices with HiFiBerry DACs (DAC+ Pro, DAC2 Pro, DAC2 HD) and none of them have this problem with HiFiBerry OS.

**HiFiBerryOS version**
20230404

**HiFiBerry sound card**
Amp2

**To Reproduce**
Steps to reproduce the behavior:
1. Start playing any content, from silence
2. Hear "pop" before playback begins
3. Stop playing the content
4. Hear another "pop"

**Expected behavior**
Playback without "pop" sounds

**Screenshots**
N/A

**Browser (if applicable)**
N/A

**Additional context**
N/A

hifiberry, 2024-03-11:
Unfortunately I can't reproduce this.

baf, 2024-03-11:
Okay - any suggestions on how I can troubleshoot?

hifiberry, 2024-03-11:
Sorry, no.

hifiberry, 2024-03-14:
Could be. Please contact support@hifiberry.com

closed 2024-03-14 by hifiberry