hifiberry/hifiberry-os #528: HifiberryOS 64 Alpha 6: ERROR: powercontroller - Couldn't import RPi.GPIO, won't load powercontroller module

baldrianbandit, 2024-04-05:
DO NOT remove the blocks below, but fill these with the requested data. Incomplete bug reports will be ignored!
You should remove this comment before posting the bug report.

**Describe the bug**
Missing python module RPi.GPIO. Powercontroller does not work.

**HiFiBerryOS version**
HifiberryOS 64 Alpha 6

**To Reproduce**
journalctl -xe | grep powercontroller

**Additional context**
Tested on RPi4

hifiberry, 2024-04-10:
e2e4d00fd983606b5bad1b9b37650a5d220598b8

Try the next HBOS release that will be releases soon.

baldrianbandit, 2024-04-12:
Tested Alpha 7: Still GPIO issues:
```
# journalctl | grep power
Apr 12 14:44:26 alarm reconfigure-players[968]:  Interrupt pin for power controller: 2 (was 2)
Apr 12 14:44:38 alarm python3[1252]: INFO: audiocontrol2 - Controller class: ac2.plugins.control.powercontroller.Powercontroller
Apr 12 14:44:38 alarm python3[1252]: INFO: powercontroller - Using controller int pin 2 on GPIO 15
Apr 12 14:44:38 alarm python3[1252]: INFO: powercontroller - found powercontroller software version 5 on I2C address 119
Apr 12 14:44:38 alarm python3[1252]: ERROR: powercontroller - Couldn't start GPIO callback, aborting... (name 'GPIO' is not defined)
Apr 12 14:44:38 alarm python3[1252]: INFO: audiocontrol2 - started controller powercontroller
Apr 12 14:44:38 alarm python3[1252]: INFO: powercontroller - Update LED state for state=unknown
Apr 12 14:44:43 alarm systemd[1]: Starting Shutdown via power controller...
Apr 12 14:44:43 alarm systemd[1]: Finished Shutdown via power controller.
```

baldrianbandit, 2024-04-21:
Will reopen for current alpha release.

closed 2024-04-21 by baldrianbandit