hifiberry/hifiberry-os #529: Airplay volume way too low

cram-cram, 2024-04-05:
**Describe the bug**
The Airplay Volume Control seems incorrect. When I'm on 50% power the volume with my Amp2 is very low, fit feels like 10%. It only gets a louder when the slider is around 70 or 80%. It doesn't make a difference whether I control the volume via my iOS device or in the web-interface. 
The volume control of Spotify Connect 50% also feels like 50% of amp power. 

**HiFiBerryOS version**
20240319

**HiFiBerry sound card**
Amp2 

**To Reproduce**
1. Play a song via Spotify Connect on iOS, set the volume to 50%
2. Play the song on Spotify iOS via Airplay, set the volume to 50%
→ drastic difference in volume

**Expected behavior**
Volume control "feels correct" on Airplay with Amp2. 50% is *not* almost inaudible on Airplay. 
50% volume via Airplay is the same as 50% volume via Spotify Connect. 

**Screenshots**
Not applicable

**Browser (if applicable)**
Not applicable

hifiberry, 2024-04-10:
Unfortunately different players handle volume control differently. As most of our sound cards like the Amp2 provide a decibel volume scale, different players map this differently to percentages. As long as a player doesn't offer specific ways to control the volume, we can't really do something about it.

cram-cram, 2024-04-11:
I downgraded to the latest stable release of Hifiberry OS, in this version the Airplay volume behaviour is perfectly appropriate, the difference is big. Only in the latest 64bit version it doesn't behave correctly. So there must have been a change in the software in between that affects volume behaviour, this is not a pure hardware issue only.

hifiberry, 2024-04-11:
Yes, there is. The Shairport version on HBOS64 is a a newer version. You can try the latest HBOS64 Alpha 7 that comes with yet another version of shairport-sync.

cram-cram, 2024-04-11:
I unfortunately don't have the time to do that now, will give it hopefully a spin later when there is time do download and install a version. I kindly ask you to give it a test drive yourself with version 20240319 and Airplay. The Airplay volume control of 80% of the amp power being only in the upper 20% of the volume scale would be a showstopper for me personally for using Hifiberry OS. That's why I thought I'd tell you :-) Thx for your support!

baldrianbandit, 2024-04-12:
I can reproduce this on Arch linux / pipewire. Most significant audible volume increase between 60% and 80%. I think shairplay scales logarithmic. Doesn't bother me, though.

Guy559, 2024-04-21:
Thank you for initiating this issue - I logged in specifically to do the same. 

As OP describes, volume control worked fine in non-64-bit version. Now, using AirPort, device volume has to be 50% before any audio can be heard. This compresses large steps in volume changes between 50-100% volume.

I see @hifiberry post above and will try the latest alpha7 prerelease. Is there a way to implement a pre-release OTA, or is it necessary to reflash the SDC? In “General” the only software update listed is from mid-January 2024.

Thanks for all. Overall, 64-bit version is an improvement. I’ve noticed I no longer have AirPort dropouts during playback which is terrific (and much needed). Thanks for all the development work.

hornej, 2025-06-21:
I just upgraded from 32bit to 64alpha8 and also experience this exact same issue