hifiberry/hifiberry-os #530: Rpi 3b+ compatablility issue with 64A1 OS

tompeel, 2024-04-10:
**Describe the bug**
Image doesent seem to boot on RPi 3b+
Using Rpi Imager to flash the image to 3b+ the green light doesent even try to flash, same board and card working well with other non 64bit HiFiBerry OS. 
This might not be a big deal as its an old board but thought it worth noting.

**HiFiBerryOS version**
HifiBerryOS 64 alpha 1

**HiFiBerry sound card**
DAC+

**To Reproduce**
Steps to reproduce the behavior:
Rpi Imager to flash SD
Power Pi on
Observe very little for a short while and re flash it a few more times just incase.

**Expected behavior**
To boot into OS

tompeel, 2024-04-10:
didnt realise we were on alpha 5, the link fomr the main bifi berry page takes you directly to A1 jsut testing A5 now but it looks better!

hifiberry, 2024-04-10:
We need to change the link. If you can wait a bit longer, there will be a new alpha version very soon with lots of changes and enhancements

closed 2024-04-10 by hifiberry

tompeel, 2024-04-10:
Ok, this can be closed. I confirm its working well on a Pi 3b+ with a DAC+ with Spotify connect specifically being tested.