hifiberry/hifiberry-os #531: HifiBerry64 Alpha 7: docker spotifyd: Failed to initialize DBus connection

baldrianbandit, 2024-04-10:
**Describe the bug**
Docker container spotifyd does not start due to dbus error due to missing DISPLAY variable

**HiFiBerryOS version**
Hbos64/Alpha7

**To Reproduce**
```
docker status spotifyd
The application panicked (crashed).
Message:  Failed to initialize DBus connection: D-Bus error: Unable to autolaunch a dbus-daemon without a $DISPLAY for X11 (org.freedesktop.DBus.Error.NotSupported)
Location: src/dbus_mpris.rs:169
```

hifiberry, 2024-04-11:
The error message isn't really correct, but it seems, the connection to the host dbus isn't working. 

Do you get the same problem after you rebootet the system?

What0s the content of /etc/dbus.1/system.d/spotifyd.conf?

baldrianbandit, 2024-04-12:
```
# cat /etc/dbus-1/system.d/spotifyd.conf
<!DOCTYPE busconfig PUBLIC "-//freedesktop//DTD D-BUS Bus Configuration 1.0//EN"
 "http://www.freedesktop.org/standards/dbus/1.0/busconfig.dtd">
<busconfig>

 <!-- Player can own the Spotify service -->
  <policy user="player">
    <allow own="org.mpris.MediaPlayer2.spotifyd"/>
    <allow own="org.mpris.MediaPlayer2.spotifyd.instance1" />
  </policy>

</busconfig>
```

baldrianbandit, 2024-04-12:
Full log reads:
```
# docker logs spotifyd
Backtrace omitted. Run with RUST_BACKTRACE=1 environment variable to display it.
Run with RUST_BACKTRACE=full to include source snippets.
Loading config from "/etc/spotifyd.conf"
No proxy specified
Using software volume controller.
Ignoring blacklisted access point ap-gew4.spotify.com:4070
Ignoring blacklisted access point ap-gew4.spotify.com:443
Ignoring blacklisted access point ap-gew4.spotify.com:80
Connecting to AP "ap-gew1.spotify.com:443"
Authenticated as "####" !
Mixing with softvol and volume control: Log(60.0)
Country: "DE"
Converting with ditherer: tpdf
Using AlsaSink with format: S16
The application panicked (crashed).
Message:  Failed to initialize DBus connection: D-Bus error: Unable to autolaunch a dbus-daemon without a $DISPLAY for X11 (org.freedesktop.DBus.Error.NotSupported)
Location: src/dbus_mpris.rs:169
```

hifiberry, 2024-04-12:
Did you login with username/password. This will only work with username/password configured.

baldrianbandit, 2024-04-12:
Yes, login data provided and logged in. Spotify is playing currently. Steps to reproduce:

1. spotify daemon playing
2. docker stop spotifyd
3. vi /etc/spotifyd.conf and set use_mpris = true
4. docker start spotifyd
5. docker los spotifyd shows dbus error

baldrianbandit, 2024-04-12:
If DBUS_SESSION_BUS_ADDRESS is unset, dbus looks for existing X11 session to find current bus address. See Automatic Launching here: https://dbus.freedesktop.org/doc/dbus-launch.1.html

hifiberry, 2024-04-15:
Unfortunately I can't reproduce it here. 

What's the output of 

 docker image ls

baldrianbandit, 2024-04-16:
Fresh install of Hbos64/alpha7 and can't reproduce either. No dbus related error message and use_mpris just works. There must have been some hickup in my last installation. Sorry for the inconvenience and thanks for your support.

closed 2024-04-16 by baldrianbandit

growers12, 2024-04-16:
Thank you for the feedback