hifiberry/hifiberry-os #533: HiFiBerryOS64 Alpha 7 - Not finding radios I used to listen in previous Alpha.

mbolo01, 2024-04-11:
DO NOT remove the blocks below, but fill these with the requested data. Incomplete bug reports will be ignored!
You should remove this comment before posting the bug report.

**Describe the bug**
In Alpha 6 (and previous versions), I could find and use many radios that I can no longer find in Alpha 7.
Some example:
Radio Paradise (US)
France Inter (France)
FIP (France)
...

**HiFiBerryOS version**
20240410

**HiFiBerry sound card**
DAC+DSP

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
Have same radio choices than in previous HBOS version 

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Browser (if applicable)**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

hifiberry, 2024-04-11:
Hmm, can you find any Radio station? The code hasn't been changed, so if you can only find fewer stations, I'm not sure where to look at.

mbolo01, 2024-04-11:
Yes I can find some stations, but the list is very short where I'm expecting more items. I've tried searching for BBC for example and I have no single results

baldrianbandit, 2024-04-12:
BBC doesn't yield for me as well. But dlf and fm4 does.

mbolo01, 2024-04-12:
looks like a more global issue to HBOS users: https://support.hifiberry.com/hc/en-us/community/posts/18138612481309

hifiberry, 2024-04-15:
The Radio module uses TuneIn as the backend and will find only stations available by TuneIn in your country. This list can change from one to to the next. We have no control over this.

mbolo01, 2024-04-15:
The issue is definitively with TuneIn then.
I booted with one  SD with HBOS Alpha4, where I have many favorites radios set,  including French stations, and that were working well. Now only the ones that can be found through the search are playable, the other favorites don't play anything while the http request seems correct.
So Radios are no longer operating correctly.
This is consistent with other HBOS users' reports.

hifiberry, 2024-04-15:
As long as nobody offers a better radio database, there's nothing we can do about it

mbolo01, 2024-04-15:
I just found that the French company owning the radios I'm listening regularly has decided to leave TuneIN on April 4th ... some other may have done the same too.
As HBOS and B&O are working close together, any chance to see HBOS Radios using what B&O uses ?

hifiberry, 2024-04-15:
No, B&O isn't working on this project anymore.

mbolo01, 2024-04-15:
I'll have to live without Radios in HBOS then, that is unfortunate. I'm closing the case.

closed 2024-04-15 by mbolo01

hifiberry, 2024-04-15:
If you know the URLs they are using, you can edit the file /etc/beocreate/radio.json

E.g. like this:

{"favourites":{"s77934":{"title":"ByteFM","img":"http://cdn-profiles.tunein.com/s77934/images/logoq.jpg?t=158748","url":"http://opml.radiotime.com/Tune.ashx?id=s77934&partnerId=no default"}}}

mbolo01, 2024-04-15:
Ah great, I was about to ask, but isn't radio.json overridden at each new version of HBOS ?