hifiberry/hifiberry-os #534: HiFiBerryOS64 - Radios not working

mbolo01, 2024-04-11:
DO NOT remove the blocks below, but fill these with the requested data. Incomplete bug reports will be ignored!
You should remove this comment before posting the bug report.

**Describe the bug**
When I managed to find a radio I like, e.g. Couleur 3 (Swiss), clicking on the icon does not start the radio.
**HiFiBerryOS version**
20240410

**HiFiBerry sound card**
DAC+DSP

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
Radios are working as expected

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Browser (if applicable)**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

mbolo01, 2024-04-11:
Correction: Available radios are working, double click on the icon was required to start a radio

mbolo01, 2024-04-11:
After closing the issue I re-opened it as the Radio behavior is inconsistent and not working ultimately

baldrianbandit, 2024-04-12:
Can't reproduce. For me, radio stations start playing after single click.

mbolo01, 2024-04-12:
Simple click is correct. The issue I faced was going back to the HiFiBerry OS radio screen after long inactivity period. Here  a single click does nothing (nor a double click I guess) unless I reload the page.  I'm on MacOS/Safari

hifiberry, 2024-04-12:
Try to open web developer tools (I don't know if Safari has such a thing and check for Javascript errors.

mbolo01, 2024-04-15:
There is no error message in  the Javascript console when I click on a radio tile, I can I see the url request for the selected radio but it takes at least 20 to 30 seconds before  any sound comes out of it, reason why I thought that a double click was required at the time I raised the issue as I could have tried simple clicks and a finally double click within the 20 sec time.

hifiberry, 2024-04-15:
Is your Plexamp container active? It might be blocking the sound card. Try disabling all containers and see if this changes something

mbolo01, 2024-04-15:
I tried different scenari and I cannot reproduce the slow radio startup anymore (where I could a couple of hours ago, but I had to reboot the RPI that was becoming non responsive. That say, Plexamp container and radio sources behave correctly IMO. I cannot start Plexamp is the radio is running, I cannot start the Radio if the Plexamp container is running, I have to stop playing a source before being able to play another. I'll try to catch the context if this happens again.

closed 2024-04-21 by hifiberry