hifiberry/hifiberry-os #540: HiFiBerryOS 64: Analogue input service fails to start

Wassup789, 2024-04-22:
## Describe the bug
On HifiBerryOS 64, the analogue Input / alsaloop service does not start or function as expected. When an attempt to enable the service is started, it instantly goes to a failed state.

I have [attached a PR](https://github.com/hifiberry/hifiberry-os/pull/539) that partially fixes this issue (see "Additional context"). However, the issue with the Python script running the audio loopback still persists.

This bug and the fixes were performed on a Raspberry Pi 5.

## HiFiBerryOS version
20240410

## HiFiBerry sound card
DAC+ ADC Pro

## Steps to reproduce the behavior:
1. Access the GUI
2. Click on "SOURCES" tab
3. Open the Analogue Input source
4. Turn on the service
5. Service should fail to start (internally)

## Expected behavior
The analogue input / alsaloop service should start and audio loopback should begin as expected.

## Additional context

### Solving the issue with the `alsaloop` service

Investigating the `journalctl` logs, the issue seems to be the `alsaloop` service failing to start.

<details>
<summary>Relevant log</summary>

```
Alsaloopwrapper thread exception: org.freedesktop.DBus.Error.AccessDenied: Connection ":1.155" is not allowed to own the service "org.mpris.MediaPlayer2.alsaloop" due to security policies in the configuration file
```
</details>

Which suggests that the D-Bus configuration file for the `alsaloop` does not exist. I created the following file in `/usr/share/dbus-1/system.d/alsaloop.conf` which resolves the `alsaloop` service failing to start.

https://github.com/Wassup789/hifiberry-os/blob/2a6fecad2a17cf23f9281eb2c2f8e0a42de2b3a9/buildroot/package/hifiberry-alsaloop/dbus.conf#L1-L13

I have submitted a PR for this fix here: https://github.com/hifiberry/hifiberry-os/pull/539

### New issue with `alsaloop.py`
Though the service starts, I encountered another error specifically with the `alsaloop.py` script that crashes when the script begins audio loopback.

<details>
<summary>Relevant log</summary>

```
Traceback (most recent call last):
  File "/opt/alsaloop/alsaloop.py", line 194, in <module>
    output_device.write(data)
alsaaudio.ALSAAudioError: Broken pipe [default]
```
</details>

This appears to be a known issue with the [pyalsaaudio](https://github.com/larsimmisch/pyalsaaudio) package (see https://github.com/larsimmisch/pyalsaaudio/issues/130), which suggests an issue with specifically v0.10.0. 

I was able to confirm that the `alsaloop.py` script (and consequently the full `alsaloop` service) works when I downgraded the version of `pyalsaaudio` to v0.9.2 (by cross-compiling the `pyalsaaudio` package and directly loading the compiled binaries from the `alsaloop.py` script).

This bug has been [reverted](https://github.com/larsimmisch/pyalsaaudio/pull/131), however, there have [not been any releases](https://pypi.org/project/pyalsaaudio/#history) that contain this fix (the latest version is still 0.10.0).

I am not sure what is the best fix for this as I was unable to figure out a way to get `alsaloop.py` to work with v0.10.0 and it seems that the other services in this repository depend on the `pyalsaaudio` package as well, but this should provide a good jump start.

hifiberry, 2024-05-17:
I have implemented some changes that seem to work her. Unfortunately I don't have an analog audio source here to test this. I recommend to try with the net alpha release when it's available (hopefully next week)

billkach, 2024-08-08:
Hi all, I'm also experiencing this issue on a RPi 4B with DAC+ ADC (not Pro). I'm running the latest build of HiFiBerryOS 64. Are there any workarounds available?

Wassup789, 2024-08-10:
@billkach, I haven't had time to test out HifiBerryOS 64 on alpha8 but I do have a working workaround for [alpha7](https://github.com/hifiberry/hifiberry-os/releases/tag/64alpha7).

Here are the steps:
1. Copy the [contents of this file](https://github.com/Wassup789/hifiberry-os/blob/a3deffddff52f890f978b3170eb128dbdba131d7/buildroot/package/hifiberry-alsaloop/dbus.conf#L1-L13) to `/etc/dbus-1/system.d/alsaloop.conf` (the file does not yet exist).
2. Cross-compile v0.9.2 of the `pyalsaaudio` library targetting `linux-aarch64` and copy the dist-packages contents to `/opt/alsaloop/pyalsaaudio/`.
   1.  The directory structure should look as follows:
        -  /opt/alsaloop/pyalsaaudio/
            - alsaaudio.cpython-311-aarch64-linux-gnu.so
            - pyalsaaudio-0.9.2.dist-info/
                - ...
   2. Here's the compiled package for your convenience: [alsaloop-0.9.2-linux-aarch64-dist-packages.zip](https://github.com/user-attachments/files/16570822/alsaloop-0.9.2-linux-aarch64-dist-packages.zip)
3. Edit `/opt/alsaloop/alsaloop.py` and add the following lines of code to line 31:
```
script_dir = os.path.dirname(os.path.abspath(__file__))
module_path = os.path.join(script_dir, 'pyalsaaudio')
sys.path.insert(0, module_path)
```
4. Restart the Analogue Input service from the Hifiberry GUI

You _could_ try this workaround out on alpha8 (I suspect you can probably skip step 1 as that fix is already implemented).

billkach, 2024-08-15:
thanks for taking the time to follow up with me about this. I just tried this with alpha8 and found it didn't work for me. I would enable the analogue input, the button would stay filled in, and then when I would navigate away from the page it would reset back to a disabled state. I'll try again with alpha7 over the weekend!

Wassup789, 2024-08-15:
It's been a while since I first set this up for my own pi, so may have forgot some steps.

Could you try running this command then restart the analogue service?
`systemctl restart dbus`

Alternatively, rebooting the pi if the above still doesn't work.