hifiberry/hifiberry-os #544: Source "Plexamp" is not working

oberstel, 2024-05-09:
**Describe the bug**
Its not possible to start / configure / use the Plexamlp source.

**HiFiBerryOS version**
20240410

**HiFiBerry sound card**
DAC+ DSP

**To Reproduce**
1. Go to 'source'
2. Click on 'plexamp'
3. Click on 'on'
4. Plexamp extension seems to be installing (takes a while) but after that, the on button turns back off :-(

**Expected behavior**
There seems to be a bug in the docker container

**Screenshots**
```
 ✔ Container plexamp  Created                                                                                                                                                                                                                         0.3s 
Attaching to plexamp
plexamp  | node:internal/modules/cjs/loader:1282
plexamp  |   return process.dlopen(module, path.toNamespacedPath(filename));
plexamp  |                  ^
plexamp  | 
plexamp  | Error: Module did not self-register: '/app/plexamp/treble/linux-arm64/treble.node'.
plexamp  |     at Object.Module._extensions..node (node:internal/modules/cjs/loader:1282:18)
plexamp  |     at Module.load (node:internal/modules/cjs/loader:1076:32)
plexamp  |     at Function.Module._load (node:internal/modules/cjs/loader:911:12)
plexamp  |     at Module.require (node:internal/modules/cjs/loader:1100:19)
plexamp  |     at require (node:internal/modules/cjs/helpers:119:18)
plexamp  |     at Object.81564 (/app/plexamp/js/index.js:2:352739)
plexamp  |     at r (/app/plexamp/js/index.js:2:3349196)
plexamp  |     at Object.2120 (/app/plexamp/js/index.js:2:347356)
plexamp  |     at r (/app/plexamp/js/index.js:2:3349196)
plexamp  |     at Object.81546 (/app/plexamp/js/index.js:2:334557) {
plexamp  |   code: 'ERR_DLOPEN_FAILED'
plexamp  | }
plexamp exited with code 0
```

mbolo01, 2024-05-10:
Assuming you already have a Plex account, and preferably a Plex Pass too, you need to manually tweak several steps to make it work the first time:
1 - After the first attempt to make it work through the HBOS UI, ssh to HBOS, go to /data/extensions/plexamp, edit Dockerfile , make sure Node 20 is used by changing the first line of the file: FROM node:20-bullseye-slim
2 - In the same file,  modify the plexamp UID from 2002 to 2000 in the following line: 
&& useradd -g 2000 -u 2000 -s /bin/bash -d /home/plexamp -G audio plexamp 
3 - Start  Plexamp extension again and it should start but wait for the Plex Token input
4 - ssh to HBOS and open a bash within plexamp docker as plexamp user: docker exec -it -u 2000 plexamp /bin/bash
5 - from here launch Plexamp manually: node js/index.js and follow the instructions to get copy/paste the Plex token
6 - reboot HBOS and you should have it working now 

Note: Sorry if there are some mistakes above, I'm an end user too and writing these steps by memory, but it will be a good start for a user documentation

oberstel, 2024-05-10:
Thank you @mbolo01 - It's working now!

As far I concern, there is now indicator in the HiFiBerry UI that Plexamp is playing something... right?
HiFiBerry and Plexamp are running side by side and can't interact among each other.

mbolo01, 2024-05-11:
HBOS UI only shows the status of the Plexamp extension/docker container and can stop/start it.
The ´Setup Plexamp' button in Sources/Plexamp simply launches Plexamp Web UI.
So no other interaction.
I usually rely only on HBOS and DAC+DSP to manage the sound, i.e. in Plexamp I have disabled loudness, preamp gain, set Plexamp volume at its max and disabled Sampling Rate

oberstel, 2024-05-11:
>I usually rely only on HBOS and DAC+DSP to manage the sound, i.e. in Plexamp I have disabled loudness, preamp gain, set Plexamp volume at its max and disabled Sampling Rate

yes, I head the same thoughts… I have a Plex / Plexamp infrastructure already and HBOS together with DAC + DSP sems to be a perfect endpoint.

closed 2024-05-11 by oberstel