hifiberry/hifiberry-os #546: HBOS 64 Alpha 8 : Search for radios whose name is made up of two or more words does not work (RadioBrowser)

mbolo01, 2024-05-24:
DO NOT remove the blocks below, but fill these with the requested data. Incomplete bug reports will be ignored!
You should remove this comment before posting the bug report.

**Describe the bug**
Search for Radio Paradise or France Inter returns Sorry, no results.
Search for Radio or France returns results where I have to find the one I'm looking for

**HiFiBerryOS version**
20240523

**HiFiBerry sound card**
DAC+DSP

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
I'm expecting to find the radio I'm looking for by entering its name, even when it includes 2 or more words

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Browser (if applicable)**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

mbolo01, 2024-05-25:
I guess it is a RadioBrowser limitation as searching on their own site has the same behavior. That said, there are more results returned from a HBOS UI, including some inactive links.

hifiberry, 2024-05-28:
The new version 0.1 should filter known broken stations and therefore return less results. We can't do anything about the search functionality. Radiobrowser only supports a single search string