hifiberry/hifiberry-os #547: Synology 216+ Music files will not load.

greyscale80, 2024-05-25:
HiFiBerry OS will not connect to Synology 216+ Fails to load Music foldrer.
HiFiBerry OS ver 20240523
DAC+ RCA
[image]
[image]
[image]
[image]

**Describe the bug**
A clear and concise description of what the bug is.

**HiFiBerryOS version**
You find this on the "About" screen. It looks like Software 20YYMMDD

**HiFiBerry sound card**
Post the exact type, e.g. DAC+ Pro instead of "DAC+".

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Browser (if applicable)**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

hifiberry, 2024-05-28:
What's the output of

 cat /etc/smbmounts.conf
 /opt/hifiberry/bin/mount-smb.py
 
 (make sure to replace passwords before posting)

TobiasDax, 2024-05-28:
I have a similar problem with my unraid system and I think I found the problem in the `mount-smb.py` file:

`# ./mount-smb.py 
//192.168.1.123/Media/iTunes/iTunes
Failed to resolve host name '192.168.1.123.local': Timeout reached
Traceback (most recent call last):
  File "/opt/hifiberry/bin/./mount-smb.py", line 33, in <module>
    ip = result.stdout.decode().split()[1].strip() if result.returncode == 0 else ""
         ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^
IndexError: list index out of range
# `

Looks like the script appends .local to the ip address which then fails

TobiasDax, 2024-05-28:
I edited the mount-smb.py file to remove the .local in line 30

```

 # Resolve .local hostnames
            host = share.split('/')[2]
            ip = ""
            resolvhost = host
            if not host.endswith(".local"):
                resolvhost = host
``` 

Running the file again gives me this output
```

# ./mount-smb.py 
//192.168.1.123/Media/iTunes/iTunes
Failed to create host name resolver: Invalid host name
Can't load /etc/samba/smb.conf - run testparm to debug it
mount -t cifs -o user=sonos,password=sonos,rw //192.168.1.123/Media/iTunes/iTunes /data/library/music/undefined-Media-4uFK9
mount error(16): Device or resource busy
Refer to the mount.cifs(8) manual page (e.g. man mount.cifs) and kernel log messages (dmesg)
mount: (hint) your fstab has been modified, but systemd still uses
       the old version; use 'systemctl daemon-reload' to reload.

``` 

But the server is successfully mounted now. So it seems to work. Hope this gets fixed in a better way than my test :)

hifiberry, 2024-05-29:
@TobiasDax : Thanks for this. We'll look into this, but I think the original reported problem is caused by something else.

cmdrdash, 2024-07-05:
I had the same problem and I fixed it by editing mount-smb.py and removing the dash "-" in the local smbmount point name

dennisfrijlink, 2024-10-24:
> I edited the mount-smb.py file to remove the .local in line 30
>
> ```
>
> �# Resolve .local hostnames
> ������������host = share.split('/')[2]
> ������������ip = ""
> ������������resolvhost = host [...]

Same for me. commtend the part of code for resolving the .local hostnames. Got following message after running `./mount-smb.py`:
```
# ./mount-smb.py
//10.0.0.50/music/
Failed to create host name resolver: Invalid host name
Can't load /etc/samba/smb.conf - run testparm to debug it
mount -t cifs -o user=<USER>,password=<PASSWORD>,rw //10.0.0.50/music/ /data/library/music/undefined-music-ISSdA
mount: (hint) your fstab has been modified, but systemd still uses
       the old version; use 'systemctl daemon-reload' to reload.
```

But it fixed the problem for scanning