hifiberry/hifiberry-os #549: HBOS 64 Alpha 8 : Radio appears as a stopped extension and returns Error when started

mbolo01, 2024-05-26:
DO NOT remove the blocks below, but fill these with the requested data. Incomplete bug reports will be ignored!
You should remove this comment before posting the bug report.

**Describe the bug**
Radio appears as a stopped extension and returns Error when started

**HiFiBerryOS version**
20240523

**HiFiBerry sound card**
DAC+DSP

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Browser (if applicable)**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

hifiberry, 2024-05-28:
There isn't really any start/stop functionality for the Radio extension. We will make sure extensions that can't be started/stopped will show this in the GUI.

mbolo01, 2024-05-28:
Perfect