hifiberry/hifiberry-os #551: Ability to save/restore Room Compensation data to be used when moving to a new system or OS version to avoid going through a new room compensation data collect

mbolo01, 2024-05-26:
DO NOT remove the blocks below, but fill these with the requested data. Incomplete feature requests will be closed.
You should remove this comment before posting the feature request.

**Is your feature request related to a problem? Please describe.**
Ability to save/restore Room Compensation data to be used when moving to a new system or OS version to avoid going through a new room compensation data collect 

**Describe the solution you'd like**
I would like to be able to restore Room Compensation data instead of going through the full process again when listening environment has not changed.

**Describe alternatives you've considered**
No other possible alternative.

**Additional context**
Add any other context or screenshots about the feature request here.

hifiberry, 2024-05-28:
When moving to a new system you can just move the SD card. I understand that updates are a problem today, but we'll have updates of HBOS64 ready soon.

mbolo01, 2024-05-28:
Thanks. Indeed the need is when there is an OS update, which is higher in Alpha/Beta situations, but still an extra step that could be avoided when new production is released.